realsenseai/librealsense #3343: Continuos handle increassing in realsense2.dll

amatabuena, 2019-02-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | 05.10.06 | 
| Operating System & Version |   Windows 10  | 
| Kernel Version (Linux Only)    |  ----                                         | 
| Platform                                 | PC  x86 |
| SDK Version                            |  2.17.0                          | 
| Language                            |  C#                          | 
| Segment			|  PC                          | 

When running C# application using D435 camera, memory and handles increase continuosly. Checking it with Windows Performance Analyzer handles increasing occurs in mfcore.dll.

[image]

This handle increassing is causing application crash when running long time.

dorodnic, 2019-02-28:
Do you see this in one of the provided examples?

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

@amatabuena Any update? Do you see the same issue with the examples?

amatabuena, 2019-03-07:
Hi, 

Sorry for not answering before, I am short on time. 

I have tested memory with 'cs-tutorial-3-processing' and I couldn't reproduce the issue. As my context is very different to provided examples I can`t validate it without changed a lot of code or coding a new example.  I will try when I have time.

Thanks.

RealSenseCustomerSupport, 2019-03-11:
----------------------------------------------

@amatabuena Did you get chance to try? Any update? Thanks!

RealSenseCustomerSupport, 2019-03-15:
----------------------------------------------

@amatabuena Any update? Thanks!

amatabuena, 2019-03-15:
Hi, 

After doing several tests I have found that handle increassing occurs when setting some properties to the depth or color sensor ( sensor.Options[Option.XXXXX].Value).

I coundn't find the exact property that is causing the issue, because it looks like some of them are affecting.

When I have more time I will try to isolate those that cause the problem.

Regards,

amatabuena, 2019-03-18:
Hi, 

I have found that the handle increassing problem is when activating error pooling on depth sensor (Option.ErrorPollingEnabled). Any suggestion about how to resolve it?

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

@dorodnic Do you have more ideas/suggestion about the issue?

amatabuena, 2019-03-22:
Hi @dorodnic 

Using depth sensor with option ErrorPollingEnabled = false the issue dissapear. Currently I have no time to find out where the problem is when ErrorPolling is active.

Regards,

closed 2019-05-05 by RealSenseCustomerSupport