realsenseai/librealsense #3344: How do I get the depth output as disparity_frame?

amaanda, 2019-02-27:
From the docs:
 _It should be noted that the RealSense sensors can also output depth directly in disparities which would remove the need for the depth-todisparity transform._ 

How do I do this? I could not find it in any examples, only this snippet of code from Issue 1462:

rs2::frameset frameset = pipeline.wait_for_frames();
rs2::depth_frame depth_frame = frameset.get_depth_frame();
rs2::disparity_transform disparity_transform( true );
rs2::disparity_frame disparity_frame = disparity_transform.process( depth_frame );

dorodnic, 2019-02-28:
Under Stereo Module > Advanced Controls > Depth Table there is Disparity Mode toggle. 
This will make the camera output disparity map from the ASIC. I don't recommend this method for two reasons:
a. Everything under Advanced Controls is ASIC-proprietary functionality, lacking in documentation and validation coverage. 
b. The resulting frame is still a `depth_frame`, the SDK has no idea that the underlying device was switched to disparity mode. Hence, further processing such as align and pointcloud will not work. What you are actually getting is disparity encoded as uint16-s. 

Using `disparity_transform` is usually the more elegant solution, and the overhead is rather small, unless you are on a really low-end platform.

ev-mp, 2019-02-28:
@amaanda ,
As rightfully pointed by @dorodnic   - relying on proprietary controls is highly not recommended. Especially given that the functionality is also provided in the SDK.

Nevertheless, in case you're determined to proceed this route - some useful info aggregated from previously discussed threads:
 - When the `disparity_shift` flag is activate, the content of the original DEPTH_Z16 buffer will be replaced with scaled disparity  according to `depth = (baseline * focal_length *32) / disparity`. I.e you'll get HW disparity with 1LSB=1/32 pixel resolution (@agrunnet comment in #2139).

[image]

 - To see how the control can be activated in C++ please follow #2229

closed 2019-03-17 by RealSenseCustomerSupport