realsenseai/librealsense #3345: Implement object detection with YOLOv3

Rahul-Venugopal, 2019-02-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { ZR300  and  D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 16)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2 }                          | 
| Language                            |  {opencv/python}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hi ,

I found a following article related to implementing Yolo with depth sensor to get depth information with the object from the following link

https://richardstechnotes.com/2018/11/15/stereolabs-zed-depth-camera-with-yolo/.

In that post it is using stereolabs ZED depth sensor and in their github page , all the required wrappers are added. 

Is it possible to do the same with Intel Realsense ? If so , is there any links or tutorials that describe the materials and steps required to do so ? 

Thanks 
Rahul

MartyG-RealSense, 2019-02-27:
Up until now, the only example I knew of regarding YOLO being used with a RealSense camera was a tutorial written for the Intel Euclid (a candy bar sized mini PC with a RealSense ZR300 camera that is still available from Intel's online Click store).

http://www.euclidcommunity.intel.com/static/tutorials/pdf/or_tutorial_final.pdf

Some research just now also found a discussion from last year in which YOLO v3 was being used with a RealSense D415.

https://github.com/AlexeyAB/darknet/issues/940#issuecomment-393538049

dorodnic, 2019-02-28:
[rs-dnn](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/dnn) and [distance_to_object.ipynb](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb) seem like good places to start. You can plug camera output to any deep-learning algorithm and use aligned depth to calculate distances and dimensions.

Rahul-Venugopal, 2019-03-02:
Thanks @MartyG-RealSense  and @dorodnic .

I really appreciate your support and these suggestions are really a good point to start with.
I will update here , if I make any progress further.

closed 2019-03-02 by Rahul-Venugopal

jinxzzz, 2019-03-13:
Hi，
Do you have any process？

lorhm, 2019-04-30:
Hi @Rahul-Venugopal,
I'm currently trying to run Yolo on the Intel Aero RTF system. Could you share your progress?