realsenseai/librealsense #3348: RealSense D415 - No Depth stream

TDVL, 2019-02-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

TDVL, 2019-02-28:
Hello,
I have another problem with RealSense D415, I don't get a depth stream. The Viewer report is : Starting Depth stream failure.
The RGB and INFRA streams are fine.
Yesterday everything was OK, today no Depth stream.
All the settings are the same as before. I have tried all the USB ports on my laptop, also, I changed the USB cable, but nothing, still no depth frame.

TDVL, 2019-02-28:
This is the screenshot of the Viewer.
[image]

TDVL, 2019-02-28:
I also tried from USB 3 port, but the problem is the same.

TDVL, 2019-02-28:
The same problem is with the D435.

RealSenseCustomerSupport, 2019-03-01:
----------------------------------------------

Hi
It's hard to believe that both D415 and D435 are failed in depth streaming. Is it possible for you to check it again on a different computer?

Thanks

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

Hi
Just want to follow up that, do you still see the issue on the cameras?

Thanks

TDVL, 2019-03-10:
Hello,
Sorry for the late response, I had some jobs to finish in last few days.
Yes, still I can't get the depth stream using the 2.18. version of the software.
I am using the 2.17. version, this version works well so far. 
I can't get the depth stream using D415 and D435 with the 2.18. in the Viewer and in the Depth Tool also. I am using Win10.
I tried to uninstall and reinstall the 2.18. few times, but still no depth stream.
But, the GUI of 2.17. version is little different, than the 2.18. For example, I can't find the "Settings" tab in upper right corner of the 2.17.? Where can I find this tab in 2.17.?
Thank you for the help.
Kind regards.

ev-mp, 2019-03-11:
@TDVL Hello,
One of the changes in v2.18 was porting certain example functionalities from core opengl to glsl.
Specifically, for some Windows10 PC with on-board the graphic card, the display driver had to be updated.  
For instance, the follow screenshot shows a non-compatible driver version that lacks some openGl extensions  

[image]

Please check whether non-GUI examples for v2.18 can run properly - e.g  `rs-depth`

TDVL, 2019-03-11:
Dear ev-mp,
Thank you for the response.
I have updated the display driver, but still no depth frame. Also, I tried the rs-depth example (the Intel Support Team member suggested me), but still I don't get the depth stream.
The 2.17. version works fine so far, but I can't find some tabs, for example the Settings tab in right upper corner (like in 2.18. version).
I don't know what is wrong. I have tested the Win7 version on other laptop (on Win7) and it worked well.
I think there is a software bug.
Best regards

RealSenseCustomerSupport, 2019-03-15:
----------------------------------------------

Hi @TDVL,

By any chance you have found a way to overcome this issue?
After taking a closer look of the RealSense viewer app, I noticed the camera was put on high-accuracy depth setting.
Could you please try the following:
Launch RealSense Viewer 2.18.1+ > More setting > settings > restore default
Then unplug and replug the camera and see if the camera generates depth with firmware default preset.
Thank you.

TDVL, 2019-03-16:
Hello,
Thank you for the response and the suggestion.
I have tried all this, but nothing. The 2.8.1. version doesn't works.
I am using now the 2.17. version, I hope there are not significant differences between 2.18.1. and 2.17.
I can't find the More setting > settings > restore default tabs in 2.17. for example.
Also, when I am using the 2.17.  Depth Tool and when I am change the resolution from 640x480 to 1280x720 the Depth Tool records me the INFRA image, not the Depth image. 
Why?
Thank you.

RealSenseCustomerSupport, 2019-03-19:
----------------------------------------------

Hi @TDVL,

> I can't find the More setting > settings > restore default tabs in 2.17. for example.

This is expected because the control was introduced in 2.18+ and not available for 2.17. By the way, have you given the latest release 2.19.1 a try and see if the problem persists?

TDVL, 2019-03-19:
Dear Support Team,

Thank you for the notification. I have downloaded the 2.19.1. version and this version works perfect so far (last 1 hour).

Thank you for all your help.

I hope, I won't have to many questions henceforward.

Kind regards,

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

You are welcome and I am glad we can help.
Closing this one for now.

TDVL, 2019-03-21:
OK, thank you.
Best regards

closed 2019-03-21 by TDVL

TDVL, 2019-03-29:
Hello Support Team,
First, I have tested yesterday the SDK 2.19.2., but the Viewer recognized the USB port as the USB 2.0., not USB 3.0. The version 2.19.1. works well, still. I don't know what is wrong with the latest version, I tried everything (cable change, unplugging etc.).

Also, I have a question about setting. I would like to use the RealSense D415 outdoors with high accuracy setting? Are there any documents about these settings or a preset JSON file? Where can I find the documents for stereo control, advanced control, depth visualisation and postprocessing settings? I need the explanation of all parameters in those setting in order to try to tune them. I can't find any documents about this. Or, If there is a document for tuning the setting for outdoor usage, I will be thankful if someone would provide me it.

Thank you in advance.

Kind regards