realsenseai/librealsense #3350: usage of rs2.log_to_file() in python

SchotmanG, 2019-02-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | 5.11.1.0 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    | 4.15.0-45-generic                                        | 
| Language                            |  python                        | 

### Issue Description
Hi, 

We are trying to extract the point-clouds from bag files. But we run in to 2 problems:

1) During extraction of certain point-clouds (different frames / bag files)  we run into a RunTimeError during `pc.calculate(depth_frame)`.

2) The error message states there is a log file but we are unable to locate this file. We saw there was a function rs2.log_to_file() but the documentation for this using the python wrapper is scarce and we can't figure out the inputs for the function..

Any help is appreciated!

Here the code:

```python

# Declare pointcloud object, for calculating pointclouds and texture mappings
    pc = rs.pointcloud()
    # We want the points object to be persistent so we can display the last cloud when a frame drops
    points = rs.points()

    real_time = False

    pipe = rs.pipeline()
    cfg = rs.config()

    cfg.enable_device_from_file(path_bag, False)

    cfg.enable_stream(rs.stream.depth, depth_settings[0], depth_settings[1], depth_settings[2], depth_settings[3])
    cfg.enable_stream(rs.stream.color, color_settings[0], color_settings[1], color_settings[2], color_settings[3])

    pipe_profile  = pipe.start(cfg)
    rs.playback(pipe_profile.get_device())
    playback = rs.playback(pipe_profile.get_device())
    playback.set_real_time(real_time)
   
    try:
        while True:

            # Wait for the next set of frames from the camera
            frames = pipe.try_wait_for_frames()
            # Fetch color and depth frames
            depth_frame = frames.get_depth_frame()
            color_frame = frames.get_color_frame()

            if not depth_frame or not color_frame:
                print("Didnt find both frames....")
                continue

            #Tell pointcloud object to map to this color frame
            pc.map_to(color_frame)

            # Generate the pointcloud and texture mappings
            points = pc.calculate(depth_frame)

            points.export_to_ply(save_path, color_frame)

            break

    except Exception as e:
       print(str(e))
    finally:
       pipe.stop() 
```

dorodnic, 2019-02-28:
Hi @SchotmanG 
This is a bug, resolved as part of #3318

SchotmanG, 2019-02-28:
Thanks for the quick response.

I pulled and build the repo using the description here https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python.

I am running my programs in a conda environment and have troubles on how import the pyrealsense library now.. 

In the guide I linked above it says:
1)  'copy the build output (realsense2.so and pyrealsense2.so) next to your script' 
2) `export PYTHONPATH=$PYTHONPATH:/path/to/script`

I could find the following files in my `/usr/local/lib/`:
- pyrealsense2 (NOT pyrealsense2.so... ?)
- librealsense2.so
- several librealsense2.so.\<versions\>
- several pybackend2.\<versions\>

Copying them next to my script and exporting the pythonpath  still results in an ImportError..

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

Hi SchotmanG,

For your question about importing pyrealsense library, you should found the python wrapper related libraries under build/wrapper/python directory.
They are the library files and some softlink files.
You should copy them to your directory for python scripts.
If you did "sudo make install" at the end of building from source code, all the necessary library files will be copied to /usr/local/lib (again library files and related softlink files).

Regarding to your original questions, I used RS 2.19.0 release source code, which contains the fixes dorodnic pointed out.
A quick test with your code snippet has no issue.
Only one note is that I used "pipe.wait_for_frames()"

Hope this can be helpful.

Cheers!

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

Hi SchotmanG,

Do you get any updates on this one?

Thanks!

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

Hi SchotmanG,

Do you still need help on this one?

Thanks!

closed 2019-04-04 by RealSenseCustomerSupport