realsenseai/librealsense #3351: Playback of bag files crashes Unity

Dealerpriest, 2019-02-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Win 10| 
| Platform                                 | PC  |
| SDK Version                            | The one embedded in the unity package | 
| Language                            |  C# | 
| Unity Wrapper version | build 2.18.1   downloaded [here](https://github.com/IntelRealSense/librealsense/releases) |
| Unity version                      |  2018.3.6f1 |

### Issue Description
Working in Unity and C#. I'm working on getting a software synchronised (i.e. without using HW sync cable) recording to bag files using multiple realsense cameras. The way I try to achieve syncronisation as of now is by calling the ```RecordDevice.Pause()``` for each camera as soon as it's started streaming. When all camera are streaming I manually trigger ```RecordDevice.Resume()``` for all connected cameras. 
The problem is that the recorded bag files crashes makes Unity crash. If I **don´t** use ```Pause()``` or ```Resume()```in any way (and just let the RecordDevice start recording, as is default behaviour) the bag files will play nicely, without causing Unity to crash. This really beats me.

Any help would be much appreciated!

Dealerpriest, 2019-02-28:
Alright!
I have now tested some more and everything points to calling ```RecordDevice.Pause()``` being the issue. If I call ```Pause()``` before stopping Unity execution it results in a bag file that crashes Unity when played back. I guess that the bag files get corrupted somehow if the ```recordDevice``` is in a paused state when the camera stops streaming?

Anyways. It seems I'm now able to control recording pause and resume for multiple cameras and have playback from the resulting bag files.

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

@Dealerpriest  So the issue resolved, right? Thanks!

RealSenseCustomerSupport, 2019-03-27:
----------------------------------------------

@Dealerpriest If no other questions for this ticket, will close it accordingly. Thanks!

closed 2019-04-03 by RealSenseCustomerSupport

manouj, 2019-06-25:
@Dealerpriest 
Hi, I am trying to get a software synchronized recording from multiple RS cameras. Can you please explain where/in which script you called the RecordDevice.Pause() and RecordDevice.Resume() functions.
Were you able to achieve synchronized recording and playback in this way?

Thanks,
Manouj