realsenseai/librealsense #3353: can I install the SDK using linux mint?

yuan0623, 2019-03-01:
For some reason, I have to use Linux mint for my project. But I also have to use realsense depth camera D435i, is there a way yo install the relsense SDK on Linux mint? I can not figure it out myself.
I've checked about this Linux mint, they say the mint is based on Ubuntu 16, I'm confused. 
Thank you in advance!

MartyG-RealSense, 2019-03-01:
There are a few different ways to install the SDK on any flavor of Linux.

You can build the SDK from source, or do a 'bypass':

https://github.com/IntelRealSense/librealsense/issues/1132#issuecomment-363572666

Or you can do a **LibUVC-Backend** installation.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md?language=en_US

dorodnic, 2019-03-02:
Hi @yuan0623 
To correct @MartyG-RealSense answer a bit - you cannot use `libuvc` method with the D435i (yet).
However, the SDK should be mostly functional with default build on recent Linux kernel versions. 
Do you know what kernel version is used by the distro you are using?

yuan0623, 2019-03-05:
> Hi @yuan0623
> To correct @MartyG-RealSense answer a bit - you cannot use `libuvc` method with the D435i (yet).
> However, the SDK should be mostly functional with default build on recent Linux kernel versions.
> Do you know what kernel version is used by the distro you are using?

Thank you so much. I've installed it and running it on the Mint!

yuan0623, 2019-03-05:
> There are a few different ways to install the SDK on any flavor of Linux.
> 
> You can build the SDK from source, or do a 'bypass':
> 
> [#1132 (comment)](https://github.com/IntelRealSense/librealsense/issues/1132#issuecomment-363572666)
> 
> Or you can do a **LibUVC-Backend** installation.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md?language=en_US

Thank you so much, it's running on the Mint!

MartyG-RealSense, 2019-03-05:
Awesome news - thanks so much for letting us know  :)

closed 2019-03-10 by dorodnic