realsenseai/librealsense #3354: modprobe: ERROR: could not insert 'videodev': Required key not available

junjun55, 2019-03-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   	Ubuntu 16.04.6 LTS      | 
| Kernel Version (Linux Only)    | 4.15.0-34-generic                                       | 
| Platform                                 | PC  lenovo G470  |
| SDK Version                            |  Intel® RealSense™ SDK 2.0 (build 2.18.1)                     | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I install it follow the https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
When I run the scripts:  ./scripts/patch-realsense-ubuntu-lts.sh
occurred follow problems:

Patched kernels modules were created successfully
Replacing videodev :
Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Required key not available
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  videodev  module was reloaded

But I think my  kernel version is supported .
uname -r
4.15.0-34-generic
How can I solve this problems? thanks.

ev-mp, 2019-03-03:
@junjun55,
The reported error may indicate that the UEFI secure boot in activated on your PC.
Please follow #2815, #1641 to disable this feature, then reapply the patches/dkms and update.

ev-mp, 2019-03-08:
@junjun55 , any update  ?

junjun55, 2019-03-11:
Yes, I have solved my question according your suggestion.I reboot PC  and (Press F2)disable the secure boot   and  run the scripts: ./scripts/patch-realsense-ubuntu-lts.sh  then  the screen print follow context:  (input y -enter y-enter until installing success)
Reversed (or previously applied) patch detected!  Assume -R? [n] y
patching file drivers/media/v4l2-core/v4l2-ioctl.c
patching file include/uapi/linux/videodev2.h
Applying realsense-metadata patch
patching file drivers/media/usb/uvc/uvc_driver.c
Reversed (or previously applied) patch detected!  Assume -R? [n] y
Hunk #1 succeeded at 2734 (offset -37 lines).
patching file drivers/media/usb/uvc/uvc_video.c
Reversed (or previously applied) patch detected!  Assume -R? [n] y
Hunk #1 succeeded at 1227 (offset 5 lines).
Hunk #2 succeeded at 1260 (offset 5 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Reversed (or previously applied) patch detected!  Assume -R? [n] y
Hunk #1 succeeded at 165 (offset -22 lines).
Hunk #2 succeeded at 186 (offset -22 lines).
Applying realsense-hid patch
patching file drivers/iio/accel/hid-sensor-accel-3d.c
Reversed (or previously applied) patch detected!  Assume -R? [n] y
patching file drivers/iio/gyro/hid-sensor-gyro-3d.c
Reversed (or previously applied) patch detected!  Assume -R? [n] y
Applying realsense-powerlinefrequency-fix patch
.....

junjun55, 2019-03-11:
> Yes, I have solved my question according your suggestion.I reboot PC and (Press F2)disable the secure boot and run the scripts: ./scripts/patch-realsense-ubuntu-lts.sh then the screen print follow [...]

> @junjun55 , any update ?
Yes, I have solved my question according your suggestion.I reboot PC and (Press F2)disable the secure boot and run the scripts: ./scripts/patch-realsense-ubuntu-lts.sh then the screen print follow context: (input y -enter y-enter until installing success)
Reversed (or previously applied) patch detected! Assume -R? [n] y
patching file drivers/media/v4l2-core/v4l2-ioctl.c
patching file include/uapi/linux/videodev2.h
Applying realsense-metadata patch
patching file drivers/media/usb/uvc/uvc_driver.c
Reversed (or previously applied) patch detected! Assume -R? [n] y
Hunk #1 succeeded at 2734 (offset -37 lines).
patching file drivers/media/usb/uvc/uvc_video.c
Reversed (or previously applied) patch detected! Assume -R? [n] y
Hunk #1 succeeded at 1227 (offset 5 lines).
Hunk #2 succeeded at 1260 (offset 5 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Reversed (or previously applied) patch detected! Assume -R? [n] y
Hunk #1 succeeded at 165 (offset -22 lines).
Hunk #2 succeeded at 186 (offset -22 lines).
Applying realsense-hid patch
patching file drivers/iio/accel/hid-sensor-accel-3d.c
Reversed (or previously applied) patch detected! Assume -R? [n] y
patching file drivers/iio/gyro/hid-sensor-gyro-3d.c
Reversed (or previously applied) patch detected! Assume -R? [n] y
Applying realsense-powerlinefrequency-fix patch
.....

junjun55, 2019-03-11:
> @junjun55 , any update ?

Yes, I have solved my question according your suggestion.I reboot PC and (Press F2)disable the secure boot and run the scripts: ./scripts/patch-realsense-ubuntu-lts.sh then the screen print follow context: (input y -enter y-enter until installing success)
Reversed (or previously applied) patch detected! Assume -R? [n] y
patching file drivers/media/v4l2-core/v4l2-ioctl.c
patching file include/uapi/linux/videodev2.h
Applying realsense-metadata patch
patching file drivers/media/usb/uvc/uvc_driver.c
Reversed (or previously applied) patch detected! Assume -R? [n] y
Hunk #1 succeeded at 2734 (offset -37 lines).
patching file drivers/media/usb/uvc/uvc_video.c
Reversed (or previously applied) patch detected! Assume -R? [n] y
Hunk #1 succeeded at 1227 (offset 5 lines).
Hunk #2 succeeded at 1260 (offset 5 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Reversed (or previously applied) patch detected! Assume -R? [n] y
Hunk #1 succeeded at 165 (offset -22 lines).
Hunk #2 succeeded at 186 (offset -22 lines).
Applying realsense-hid patch
patching file drivers/iio/accel/hid-sensor-accel-3d.c
Reversed (or previously applied) patch detected! Assume -R? [n] y
patching file drivers/iio/gyro/hid-sensor-gyro-3d.c
Reversed (or previously applied) patch detected! Assume -R? [n] y
Applying realsense-powerlinefrequency-fix patch
.....

ev-mp, 2019-03-11:
@junjun55,
This logs suggests that you're  applying patches when these were already applied.
This may occur if git repo has been corrupted, or the patches scripts has been aborted.

Please try the following methods:
1 Browse to the `ubuntu-xenial-<your_ubuntu_flavor>` folder that should be found in the librealsense's root dir.
2. Run `git reset --hard` and `git pull` to make it up-to-date.
3. Rerun the patches script.

In case the above still fails, then rename the `ubuntu-xenial-<...>` folder and rerun the patches script

ev-mp, 2019-03-19:
Hi @junjun55 , is this has been resolved for you ?

junjun55, 2019-03-22:
> Hi @junjun55 , is this has been resolved for you ?

Sorry,  my expression is not clear. My last reply means  I  had solved the question  after  enter  y .....In the end ，it was installed successfully.  
  the screen print follow context: (that is not error)
Reversed (or previously applied) patch detected! Assume -R? [n] y
patching file drivers/media/v4l2-core/v4l2-ioctl.c
patching file include/uapi/linux/videodev2.h
Applying realsense-metadata patch
patching file drivers/media/usb/uvc/uvc_driver.c
Reversed (or previously applied) patch detected! Assume -R? [n] y
Hunk #1 succeeded at 2734 (offset -37 lines).
patching file drivers/media/usb/uvc/uvc_video.c
Reversed (or previously applied) patch detected! Assume -R? [n] y
Hunk #1 succeeded at 1227 (offset 5 lines).
Hunk #2 succeeded at 1260 (offset 5 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Reversed (or previously applied) patch detected! Assume -R? [n] y
Hunk #1 succeeded at 165 (offset -22 lines).
Hunk #2 succeeded at 186 (offset -22 lines).
Applying realsense-hid patch
patching file drivers/iio/accel/hid-sensor-accel-3d.c
Reversed (or previously applied) patch detected! Assume -R? [n] y
patching file drivers/iio/gyro/hid-sensor-gyro-3d.c
Reversed (or previously applied) patch detected! Assume -R? [n] y
Applying realsense-powerlinefrequency-fix patch

ev-mp, 2019-03-24:
@junjun55 , In your response you seem to repost the original message but do not provide any new info - did you follow the instructions suggested above, performed reset/rename of the `ubuntu-xenial-<...>` folder and installed the patches successfully ? 

Please elaborate the actual steps performed and their outcome.

RealSenseCustomerSupport, 2019-04-03:
----------------------------------------------

@junjun55 Is the issue resolved? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

@junjun55 Any update?

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@junjun55 If no other questions, will close the ticket accordingly. Thanks!

closed 2019-05-03 by RealSenseCustomerSupport