realsenseai/librealsense #3355: Exception: Access violation - no RTTI data

Dealerpriest, 2019-03-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Win 10| 
| Platform                                 | PC  |
| SDK Version                            | The one embedded in the unity package | 
| Language                            |  C# | 
| Unity Wrapper version | build 2.18.1   downloaded [here](https://github.com/IntelRealSense/librealsense/releases) |
| Unity version                      |  2018.3.6f1 |

### Issue Description
I get 
```
Exception: Access violation - no RTTI data!
rs2_playback_device_pause(device:0000000042EDD020)
```
when I try to control the playback of a bag file. I have created a playback device by calling ```playBack = PlaybackDevice.FromDevice(ActiveProfile.Device)```, but when I try to call ```Pause()``` or ```Resume()``` I get the exception above.
 What does this error mean, and how can I fix it?

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

Hi Dealerpriest,

Sorry for taking a while to come over to this issue.
Wonder if you have any update on this one? And any change if using latest LRS release?
If it persists, would you please share a simplified test code to help replicate & debug it?

Thanks!

RealSenseCustomerSupport, 2019-04-24:
----------------------------------------------

Hi Dealerpriest,

Any update? If nothing else is needed, this one will be closed soon.

Thanks!

closed 2019-05-03 by Dealerpriest