realsenseai/librealsense #3356: D435 rs_rghb.launch error hector_pose_estimation_nodlets.xml not found

nitin5, 2019-03-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

For camera  D435, Firmware version 05.09.02, Ubuntu 16.04, ROS kinetic 

On launching the rs_rgbd.launch file, an ERROR gets printed on the screen 
/camera /realsense2_camera_manager_map<std::__css_11::basic_string<char>, pluginlib::ClassDesc> pluginlib::ClassLoader<T>::determineAvailableClasses: Skipped loading plugin with error: XML Document '/opt/ros/kinetic/share/hector_pose_estimation/hector_pose_estimation_nodlets.xml' has no Root Element. This likely means the XML is malformed or missing..."

I do not have this file in the system. Is this file necessary for proper functioning of the node. I can see output in the viewer.

rgds,
Nitin

RealSenseCustomerSupport, 2019-03-08:
----------------------------------------------

Hi nitin5,

Can you please help clarify your ROS wrapper version? And also RealSense SDK version?
And again, you can stream depth/IR/RGB using RS Viewer, but can't start ROS?

On the other side, you may consider to upgrade FW to be a newer version than 5.9.2 to get better usage of FW support.

Thanks!

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

Hi nitin5,

Wonder if there is any updates?

Thanks!

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Hi nitin5,

If there is nothing else, this one will be closed soon.

Thanks!

closed 2019-03-30 by RealSenseCustomerSupport