realsenseai/librealsense #3357: Is there an API/function for toggling repeat of playback from bag files

Dealerpriest, 2019-03-01:
I'm using Unity and C#.
When creating a device from a bag file, it's possible to control playback by getting a playback device:
```
var playback = PlaybackDevice.FromDevice(ActiveProfile.Device)
```
With functions such as ```Pause(), Resume and Seek()``` I can control various aspects of the playback.
But... By default the playback repeats. Is there a way to turn this off? I know this functionality is exists in the intel realsense viewer, where I can press a button to toggle repeat on or off. Is there a builtin functionality for this in the API? Or. Do I have have to create it myself? And if so, what would you suggest be the best way to go about it?

Thankful for any assistance!

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

@Dealerpriest Sorry we don't have such API yet. You can try to create it by yourself. Thanks!

RealSenseCustomerSupport, 2019-03-27:
----------------------------------------------

@Dealerpriest Any other questions for this ticket?

RealSenseCustomerSupport, 2019-04-01:
----------------------------------------------

@Dealerpriest Any other questions about this ticket?

RealSenseCustomerSupport, 2019-04-04:
----------------------------------------------

@Dealerpriest If no other questions, will close the ticket at this point. Thanks!

closed 2019-04-11 by RealSenseCustomerSupport