realsenseai/librealsense #3358: D435 (matlab) change laser power and exposure time

aneim, 2019-03-01:
Hello, I'm using D435 with matlab wrapper.
i have long tried to figure that out, but still don't know how to turn off/on the IR Projector or how to change its Laser Power manually which should be possible according to the data sheet. Same goes for controlling the exposure time.
Can anyone tell me what the coding for that would look like?
thanks in advance!

MartyG-RealSense, 2019-03-01:
Code used with the MATLAB wrapper is said to be almost 1:1 the same as C++ code.  So you should be able to adapt C++ examples for the wrapper.

The link below may be of use to you.

https://github.com/IntelRealSense/librealsense/issues/1663

aneim, 2019-03-06:
Thank you for your answer. Sadly i still did not manage to figure out the correct syntax. Working with the mex file implemented in the matlab wrapper is kind of confusing for me so its basically try and error. I would therefore appreciate some further tips very much.
Thank you very much in advance

MartyG-RealSense, 2019-03-06:
I do not have any further information on this subject unfortunately.  Hopefully one of the Intel guys on this forum can provide help.  Good luck!

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi aneim,

You can do something like following:

    profile = pipe.start();
    dev = profile.get_device();
    d_sensor = dev.first('depth_sensor');
    disp(d_sensor.is('depth_sensor'));
    if d_sensor.supports_option(optionemitter_enabled)
        d_sensor.set_option(option.emitter_enabled, <_**0 OR 1**_>);
    end

And then similarly for changing exposure time (with supported options here: https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/option.m)

Thanks!

RealSenseCustomerSupport, 2019-03-26:
----------------------------------------------

Hi aneim,

Wonder if you have any update?
If nothing else is needed, this one will be closed out soon.

Thanks!

aneim, 2019-03-26:
Hello.
Thank you very much and sorry for my delayed answer. That one solved it for me. 
Thanks again,
aneim

RealSenseCustomerSupport, 2019-03-26:
----------------------------------------------

Hi aneim,

Thanks for the update. Glad that it works for you.
Closing this one.

Thanks!

closed 2019-04-02 by RealSenseCustomerSupport

danielesser1, 2023-09-01:
I tried using this code to change the options for a D405 camera in matlab, however it seems that only a few of the options are supported. Why can't I change options such as filter_option, accuracy, sensor_mode...

These options are available in realsense viewer, so it seems that the D405 camera supports them.

MartyG-RealSense, 2023-09-02:
The MATLAB wrapper's programming language is almost 1:1 with the SDK's C++ language, so if an option is supported for the D405 then it should be possible to access it through the MATLAB wrapper.  A list of the wrapper's options can be found at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/option.m

https://github.com/IntelRealSense/librealsense/issues/9839#issuecomment-939332084 has an example of setting options in the MATLAB wrapper.