realsenseai/librealsense #3360: PlaybackDevice returns incorrect number

Dealerpriest, 2019-03-01:
I'm using the C# wrapper for Unity. I've created a `PlaybackDevice` by calling

```
using(var playback = PlaybackDevice.FromDevice(devices[0].device)) {
	print("pos: " + playback.Position);
	print("dur: " + playback.Duration);
	print((double) (playback.Position) / (double)(playback.Duration));
}
```
for a bag file that is a few seconds long, `print("dur: " + playback.Duration);` will output `dur: 1266874888984165558` to the console. The documentaion specifies that the number should be in nanoseconds, but this number is waaaay too big for that.

Is this a bug?

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

@Dealerpriest Did you still get the issue? Could you please tell the librealsense version you get the issue? Thanks!

Dealerpriest, 2019-03-18:
v2.18.1
This one: https://github.com/IntelRealSense/librealsense/releases/tag/v2.18.1

Using the unity-package provided with that release.

Dealerpriest, 2019-04-18:
@RealSenseCustomerSupport 
I can now confirm that this happens if you pause a recorder immediately after starting the pipe. I have tried it with the SDK in C++.

Dealerpriest, 2019-04-18:
You cand confirm this by adding one line to the record playback example:
In the following snippet the line `device.as<rs2::recorder>().pause();` is the only addition to the example provided by the SDK.
```
// If it is the start of a new recording (device is not a recorder yet)
if (!device.as<rs2::recorder>())
{
	pipe->stop(); // Stop the pipeline with the default configuration
	pipe = std::make_shared<rs2::pipeline>();
	rs2::config cfg; // Declare a new configuration
	cfg.enable_record_to_file("a.bag");
	pipe->start(cfg); //File will be opened at this point
	device = pipe->get_active_profile().get_device();
	device.as<rs2::recorder>().pause();
}
```

Dealerpriest, 2019-05-02:
@RealSenseCustomerSupport 
Hello! I strongly believe this is an important bug that should receive your attention. Can you confirm whether you have acknowledged the existence of this bug?

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

I guess it should be something wrong in the flow.
Please try this snippet.

    pipeline = new Pipeline();var cfg = new Config();cfg.EnableDeviceFromFile("PATH_TO_A_BAG_FILE");var pp = pipeline.Start(cfg);using (var playback = PlaybackDevice.FromDevice(pp.Device)){ print("pos: " + playback.Position); print("dur: " + playback.Duration); print((double) (playback.Position) / (double)(playback.Duration));}

RealSenseCustomerSupport, 2020-02-26:
----------------------------------------------

@Dealerpriest Did you get chance to try? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@Dealerpriest  Will close it if no other questions about this. Thanks!

closed 2020-04-08 by RealSenseCustomerSupport