realsenseai/librealsense #3361: T265 will not open in realsense-viewer

bperseghetti, 2019-03-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | Version Installed when Shipped (won't open in viewer) | 
| Operating System & Version |  Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-45-generic | 
| Platform                                 |  Dell Precision 7510 (i7-6920HQ)  and Hades Canyon NUC8i7HVK  |
| SDK Version                            |  v2.18.1                          | 
| Language                            |  N/A | 
| Segment			|  Robot/others                           | 

### Issue Description
We are unable to open the T265 in realsense-viewer, it results in this message in the terminal:
`01/03 15:43:52,382 WARNING [139827475371776] (types.cpp:57) set_pu(id=3) failed! Last Error: Input/output error
 01/03 15:43:52,382 WARNING [139827475371776] (sensor.cpp:725) Exception was thrown when inspecting properties of a sensor`
when the realsense-viewer software is run on the NUC as well as laptop. Doesn't matter what port it is plugged into, still results in same issue. 
When I `lsusb` it shows:
`Bus 001 Device 009: ID 03e7:2150`

Interestingly no `Intel Corp.` name following it like there is with D435 and the D435 works great in realsense-viewer on both computers.
(`lsusb` for D435 shows `Bus 002 Device 002: ID 8086:0b07 Intel Corp.` )

Has anyone else had issues with their T265? @dorodnic  I know you helped out with our issue on the D435i (fixed in v2.18.1) in realsense-viewer, was curious if you have seen anything like this with T265 yet.

bperseghetti, 2019-03-01:
Does seem to work in Windows 10 Pro Version 1809 (OS Build 17763.316) with SDK v2.18.1 though, was it maybe not tested in Linux fully yet? Sadly using Windows is not an option as the system must be implemented in Linux. But at least this shows the device works, just not in a way useful to our needs.
[image]

dorodnic, 2019-03-02:
Hi @bperseghetti 
Everything should be supported on Linux as well. 
Maybe udev rules are missing? (any Permission Denied errors in the output?)
Another questions is how did you install the SDK? Is this from apt or did you build it from source?
When building the SDK from source, we require internet connection to download and embed T265 firmware into the binary (since the module does not come with firmware on board) 
Sometimes when behind proxy, CMake will fail to connect and print something to the extent of building without T265. Apt version should be good.

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

Hi @bperseghetti,

You can try to use the prebuilt dkms and realsense-viewer app (refer to below link) to run T265. If the T265 firmware (ex: 0.0.18.4577) can be loaded correctly from realsense-viewer, you should see a renewed VID:PID as 8087:0b37.
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

eclipse0922, 2019-03-04:
ubuntu 18.04, 
sdk 2.18.1
realsense viewer wit t265 works perfectly

bperseghetti, 2019-03-04:
@dorodnic thank you for your very fast reply, sorry I wasn't able to get back with you as fast. I had installed from source but grep of build didn't show any related issues/warns. I followed the advice to install from apt but first made sure to `sudo make uninstall` and remove all the files installed from source build followed by `rm -rf` the repo. I then followed the installation instructions for 16.04 from apt and it all installed without any errors or issues, furthermore I installed the `-dbg` and `-dev` packages just to play it safe and then restarted my machine (to play it extra safe). It appears that it has not helped as running `realsense-viewer` results in:

`04/03 16:14:17,315 WARNING [140486059050752] (types.cpp:57) set_pu(id=3) failed! Last Error: Input/output error
 04/03 16:14:17,315 WARNING [140486059050752] (sensor.cpp:725) Exception was thrown when inspecting properties of a sensor`

@RealSenseCustomerSupport weirdly my VID:PID from `lsusb` for the device changed to:
`Bus 001 Device 007: ID 03e7:2150`

When I run `which realsense-viewer` I get:
`/usr/bin/realsense-viewer`

RealSenseCustomerSupport, 2019-03-05:
----------------------------------------------

Hi @bperseghetti,

Suggest to refer to distribution_linux.md doc to install the prebuilt apps. ($ sudo apt-get install librealsense2-utils) And, see how the prebuilt realsense-viewer works.
You may not successfully build TM2 (BUILD_WITH_TM2 is ON). Or, probably, TM2 firmware is not downloaded successfully. You can also check if target.mvcmd file in /third-party/libtm/resources.

dorodnic, 2019-03-05:
The good news is that T265 is detected on your system (03e7:2150)
Did you try to run `rs-enumerate-devices`?

jattiogbe, 2019-03-05:
Hello@bperseghetti,

Did you manage to export the displayed trajectory to any kind of file ? Could you please share it ? 

I don't know yet if the device would be useful for my project, so I would like to see what data I get precisely and in what format.

bperseghetti, 2019-03-05:
@dorodnic running `rs-enumerate-devices`:
     `No device detected. Is it plugged in?`
 `lsusb` still shows:
     `Bus 001 Device 006: ID 03e7:2150`
`modinfo uvcvideo | grep "version:"` results in:
    `version:        1.1.2.realsense-1.3.4
     srcversion:     C4FB5E9B212E9C39ED6CD7C`

@RealSenseCustomerSupport that is the directions that I followed for installing it (through apt) to play it safe I also followed instructions to remove/purge all realsense dpkg for a reinstall just in case: 
    `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`
followed by removing the apt repo:
    `sudo add-apt-repository -r "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" -u`

and reinstalled by: 
1. `sudo apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE`

2. `sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" -u`

3. `sudo apt update` 

4. `sudo apt-get install librealsense2-dkms`

5. `sudo apt-get install librealsense2-utils`

6. `sudo apt-get install librealsense2-dev`

7. `sudo apt-get install librealsense2-dbg`

8. `sudo shutdown -r now`

Resulted in exact same issues where it is not running correctly in realsense-viewer with same warns I have posted before and same exact lsusb, rs-enumerate-devices and uvcvideo version.

@jattiogbe I'm only using Linux for my use applications but soon as I get it running correctly in Linux will upload examples of it running and data output stream.

jattiogbe, 2019-03-05:
Thanks for the info @bperseghetti . 
I wish you good luck on your project :)

chunyang-zhang, 2019-03-06:
@bperseghetti, I encounter the same case with my machine. If you have builded SDK with TM2 (BUILD_WITH_TM2 is ON) successfully, you can run command: $  sudo realsense-viewer  in your OS terminal, it works in my machine.

RealSenseCustomerSupport, 2019-03-06:
----------------------------------------------

Hello
chunyang-zhang

dorodnic, 2019-03-06:
@chunyang-zhang - If `sudo` solved the issue for you, you should be able to do:
```sh
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/ 
sudo udevadm control --reload-rules && udevadm trigger
sudo rm ~/.realsense-config.json
```
And then run the SDK tools without `sudo`

chunyang-zhang, 2019-03-06:
@dorodnic  thanks a lot ! It runs very well without sudo

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

Hi @bperseghetti,

Do you have any good progress? similar issue with @chunyang-zhang?

antmusco, 2019-03-07:
@RealSenseCustomerSupport I'm having the same issues as @bperseghetti, running Ubuntu 16.04.

I've tried installing from apt repositories and building from source (both v2.19.0)

The binaries rs-capture and rs-pose both work fine.

I noticed that I can get the USB ID to update to the correct value of 0b37 by running `rs-pose`, but I'm not sure that the firmware is being updated since it is not persistent. Here are the steps to reproduce:
1. Plug T265 into USB3 port
2. `lsusb` outputs `ID 03e7:2150 _ `
3. run `rs-pose`
4. `lsusb` outputs `ID 8087:0b37 Intel Corp.`.
5. `realsense-viewer` outputs (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor`
6. `rs-enumerate-devices` outputs `No device detected. Is it plugged in?`
7. Unplug/Replug the T265
8. `lsusb` outputs `ID 03e7:2150 _ `

RealSenseCustomerSupport, 2019-03-08:
----------------------------------------------

Hi @antmusco,

1. Any more dmesg and librealsense logs beteen step 3 and step 5?
2. Indeed, "rs-enumerate-devices" can't work for T265, need to be addressed.

RealSenseCustomerSupport, 2019-03-08:
----------------------------------------------

Hi @antmusco,

Please also run "lsusb -v" with more usb info.

antmusco, 2019-03-08:
Here is the `dmesg` output and `lsusb -v` immediately after the associated steps:

1. Plug T265 into USB3 port
```
[ 3088.626937] usb 1-4: new high-speed USB device number 6 using xhci_hcd
[ 3088.775782] usb 1-4: New USB device found, idVendor=03e7, idProduct=2150
[ 3088.775788] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3088.775792] usb 1-4: Product: Movidius MA2X5X
[ 3088.775796] usb 1-4: Manufacturer: Movidius Ltd.
[ 3088.775799] usb 1-4: SerialNumber: 03e72150
```
[lsusb_2150.txt](https://github.com/IntelRealSense/librealsense/files/2946484/lsusb_2150.txt)

2. run `rs-pose`
```
[ 3144.708740] usb 1-4: USB disconnect, device number 6
[ 3145.230738] usb 2-5: new SuperSpeed USB device number 3 using xhci_hcd
[ 3145.251843] usb 2-5: New USB device found, idVendor=8087, idProduct=0b37
[ 3145.251849] usb 2-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3145.251852] usb 2-5: Product: Intel(R) RealSense(TM) Tracking Camera T265
[ 3145.251856] usb 2-5: Manufacturer: Intel(R) Corporation
[ 3145.251859] usb 2-5: SerialNumber: 8454121116760000
```
[lsusb_0b37.txt](https://github.com/IntelRealSense/librealsense/files/2946485/lsusb_0b37.txt)

bperseghetti, 2019-03-08:
@dorodnic @RealSenseCustomerSupport @chunyang-zhang  Thanks for the help, still having issues with it on laptop, I think it might have to do with interference of the built-in webcam. I will have a longer update of all my findings between 6 different systems, just been swamped with other more pressing work to give a quality response. As a side note @antmusco do you have a built in camera on the system you are trying to run it on? I found that switching to a system that doesn't have one lets the T265 work. Or at least it worked on my Desktop with 16.04 and using v2.18.1. I will update more info on all the different issues/solutions I found probably over the weekend or maybe on Monday, just didn't want people to think that I was ignoring the issue/thread.

antmusco, 2019-03-08:
@bperseghetti I'm running on a laptop with a camera, yes. I'm able to use the API successfully; only realsense-viewer is having the issue.

RealSenseCustomerSupport, 2019-03-10:
----------------------------------------------

Hi @antmusco, @bperseghetti,

Several steps for further check:
1. re-plug T265
2. run "$ sudo udevadm monitor"
3. enable librealsense log file "export LRS_LOG_LEVEL=DEBUG" and launch realsense-viewer in another terminal
4. show the log in that terminal running above udevadm monitor command. Like to check if any expected uevents
5. show the librealsense log file

If you can dump the usb traffic log from usbmon. That's also great.

zhuoyueljl, 2019-03-12:
> The good news is that T265 is detected on your system (03e7:2150)
> Did you try to run `rs-enumerate-devices`?
when i run rs-enumerate-devices on terminal with T265, terminal show No device detected. Is it plugged in .  Do you have any suggestion?

antmusco, 2019-03-12:
@RealSenseCustomerSupport

For the following logs, I unplugged the T265, began recording logs, plugged in the T265, and ran `realsense-viewer` in another terminal with the indicated DEBUG log setting.

Here's the output from the udevadmin monitor: 
[udev_monitor.txt](https://github.com/IntelRealSense/librealsense/files/2957654/udev_monitor.txt)

Here's the stream output from usbmon:
[usbmon_stream.txt](https://github.com/IntelRealSense/librealsense/files/2957649/usbmon_stream.txt)

Here's the output using the DEBUG log setting (nothing seems to be new)

```bash
$ export LRS_LOG_LEVEL=DEBUG
$ realsense-viewer 
 12/03 11:26:28,019 WARNING [139677941581568] (types.cpp:59) set_pu(id=3) failed! Last Error: Input/output error
 12/03 11:26:28,019 WARNING [139677941581568] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor
 12/03 11:26:33,058 WARNING [139677941581568] (types.cpp:59) set_pu(id=9) failed! Last Error: Input/output error
 12/03 11:26:33,058 WARNING [139677941581568] (sensor.cpp:753) Exception was thrown when inspecting properties of a sensor
```

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

Hi @antmusco,

1. Per udev_monitor.txt and usbmon_stream.txt, I don't see the event for T265. Probably, you start the monitor after T265 connected. If so, it looks the T265 firmware is not loaded correctly, I don't see usb disconnect event and T265 usb device re-enumeration.
2. From librs log, the warning should be due to another non-realsense USB camera.
Is it feasible to disable the non-realsense USB camera and check if the realsense-viewer can recognize T265?

zlacelle, 2019-03-14:
Adding to this thread that I have the same issue. The errors I'm seeing:
 14/03 14:21:10,207 WARNING [140300537014016] (types.cpp:57) set_pu(id=3) failed! Last Error: Input/output error
 14/03 14:21:10,207 WARNING [140300537014016] (sensor.cpp:715) Exception was thrown when inspecting properties of a sensor
 14/03 14:21:15,314 WARNING [140300537014016] (types.cpp:57) set_pu(id=9) failed! Last Error: Input/output error
 14/03 14:21:15,314 WARNING [140300537014016] (sensor.cpp:715) Exception was thrown when inspecting properties of a sensor

This is following instructions to install all packages from the repo, not compiling locally. Ubuntu 16.04, kernel 4.15.

rs-enumerate-devices:
 Device info: 
    Name                          : 	Platform Camera
    Serial Number                 : 	1-7-4
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/video4linux/video0
    Product Id                    : 	58C2

Stream Profiles supported by RGB Camera
 Supported modes:                                      stream     resolution fps   format   
    Color	  640x480	@ 30Hz	RGB8
    Color	  640x480	@ 30Hz	Y16
    Color	  640x480	@ 30Hz	BGRA8
...

 modinfo uvcvideo | grep "version:"
version:        1.1.2.realsense-1.3.4
srcversion:     C4FB5E9B212E9C39ED6CD7C

RealSenseCustomerSupport, 2019-03-15:
----------------------------------------------

Hi @zlacelle, @zhuoyueljl,

You can try below workaround to let rs-enumerate-devices work first.
https://github.com/IntelRealSense/librealsense/pull/3439

@zlacelle,
Per your test with the same error, realsense-viewer doesn't work. rs-pose works, right?

zlacelle, 2019-03-15:
@RealSenseCustomerSupport Yes, rs-pose works.

Also, my mistake, I had an older version (2.13) of librealsense2 in /usr/local from using the other cameras before, and hadn't removed it. Removing it makes the realsense-viewer work.

However, the printouts are still there FWIW.

RealSenseCustomerSupport, 2019-05-03:
----------------------------------------------

Since this is fixed, if not further question, will close this.

closed 2019-05-10 by RealSenseCustomerSupport

victorll998, 2021-06-03:
@RealSenseCustomerSupport 

Hi there, I'm experiencing the same issue of t265 "no device detected. Is it plugged in?" I tried the aggressive method #3439 to enumerate device but t265 still won't show.

This is happened when I downgraded SDK version from 2.45.0 to 2.42.0. To avoid other version of librealsense2 residual after uninstall I had a fresh reinstall of Ubuntu 20.04. Then I installed the 2.42.0 from source according to [this guide] (https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). The result is that I can run my d455 without any issue but cannot detect t265.

I got a similar output from usbmsg. Please reopen this issue and take a look into my case.

```
[ 3295.571179] usb 2-3: USB disconnect, device number 2
[ 3807.580753] usb 3-3: USB disconnect, device number 10
[ 3809.726868] usb 3-3: new high-speed USB device number 11 using xhci_hcd
[ 3809.876183] usb 3-3: New USB device found, idVendor=03e7, idProduct=2150, bcdDevice= 0.01
[ 3809.876186] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3809.876189] usb 3-3: Product: Movidius MA2X5X
[ 3809.876190] usb 3-3: Manufacturer: Movidius Ltd.
[ 3809.876192] usb 3-3: SerialNumber: 03e72150
```
Device: Intel NUC11
Kernel: 5.8.0-43-generic
SDK: 2.42.0