realsenseai/librealsense #3362: Focus on SR300

AZfrima, 2019-03-01:
Hello,

It seems like the focus is not supported by the library. Am I wrong ?
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                               
|---------------------------------|------------------------------------------- 
| Camera Model                       | SR300 | 
| Firmware Version                   | 
| Operating System & Version |   {Win (8.1/10)   | 
| Kernel Version (Linux Only)    |                                    
| Platform                                 | PC  
| SDK Version                            |  pyrealsense2                          
| Language                                |  python                         
| Segment			          | AR                            

### Issue Description

dorodnic, 2019-03-02:
Hi @AZfrima 
You are correct. More precisely controlling the focus is not supported by the SR300 device.

AZfrima, 2019-03-06:
Thanks dorodnic :)

closed 2019-03-07 by AZfrima