realsenseai/librealsense #3363: Inverted Color on Depth images

Xiphosphere, 2019-03-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 and D415 | 
| Firmware Version                   | 5.11.1 | 
| Operating System & Version |   Ubuntu Mate 16.04.2-armhf  | 
| Kernel Version (Linux Only)    |  4.14.94-v7+                                        | 
| Platform                                 | Raspberry Pi 3 Model B via USB 2.1 |
| SDK Version                            |  2.17                          | 
| Language                            |  Python 2.7 / OpenCV 3.4                       | 

### Issue Description

[image]

Tried applying colorizer to the depth image but colors are inverted where red is near and blue is far.
New to RealSense and programming as a whole, so do need some pointers in the right direction.

Code attached is a modified version of the opencv_viewer_example.py

[rs.txt](https://github.com/IntelRealSense/librealsense/files/2921861/rs.txt)

import pyrealsense2 as rs
import numpy as np
import cv2

pipeline=rs.pipeline()
config=rs.config()

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 15)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 15)
config.enable_stream(rs.stream.infrared, 640, 480, rs.format.y8, 15)

cfg = pipeline.start(config)
dev = cfg.get_device()

colorizer = rs.colorizer()
colorizer.set_option(rs.option.color_scheme, 0) 

depth_sensor = dev.first_depth_sensor()
depth_sensor.set_option(rs.option.visual_preset, 2)

try:
    while True:

          frames = pipeline.wait_for_frames()
          depth_frame = frames.get_depth_frame()
          color_frame = frames.get_color_frame()
          ir1_frame = frames.get_infrared_frame(1)
          if not depth_frame or not color_frame or not ir1_frame:
               continue

          depth_image = np.asanyarray(colorizer.colorize(depth_frame).get_data())
          color_image = np.asanyarray(color_frame.get_data())
          ir1_image = np.asanyarray(ir1_frame.get_data())

          cv2.namedWindow('RGB_RealSense', cv2.WINDOW_AUTOSIZE)
          cv2.namedWindow('Depth_RealSense', cv2.WINDOW_AUTOSIZE)
          cv2.namedWindow('IR_RealSense', cv2.WINDOW_AUTOSIZE)
          cv2.imshow('RGB_RealSense', color_image)
          cv2.imshow('Depth_Realsense', depth_image)
          cv2.imshow('IR_RealSense', ir1_image)
          key = cv2.waitKey(1) & 0xFF
          if key == ord("q"):
               break

finally:

         pipeline.stop()

RealSenseCustomerSupport, 2019-03-05:
----------------------------------------------

@Xiphosphere
For color scheme, you can refer to https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_processing.hpp#L609

For visual preset, could you elaborate more about "unable to successfully change"? Do you mean you can't see any change with visual preset or you got any error? Thanks!

Xiphosphere, 2019-03-05:
Hi, thanks for the prompt reply, but i am unable to see any changes in the visual preset when i try to change it via the python code. However, the visual presets do change as per normal when running the realsense-viewer.

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

@Xiphosphere Sorry for late response! I can see the difference from visual preset 5 and preset 2. What values did you test?  I'll attach the snapshot of my testing.

sherrysun2018, 2019-03-20:
@Xiphosphere  Please see the snapshot of visual preset 5 as below.
[image]

And below is the snapshot of visual preset 2.
[image]

They showed some difference.

RealSenseCustomerSupport, 2019-04-01:
----------------------------------------------

@Xiphosphere  If no other questions, will close the ticket accordingly. Thanks!

Xiphosphere, 2019-04-01:
Hi apologies for the super late responses due to other tests being conducted, just tested on another Raspberry Pi and there has been a difference in the visual preset. Will close this ticket. Thanks again for all the hard work!

closed 2019-04-01 by Xiphosphere

DuyguSerbes, 2021-05-06:
Your code you provided is very beneficial. Regarding the laser power, I have a question. How can I determine IR power on python script? I can arrange it via Realsense Viewer, but I dont know is there a way for python script?