realsenseai/librealsense #3364: How to get rgb value from color frame

TouchDeeper, 2019-03-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   |05.11.01.00| 
| Operating System & Version |    Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0                                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.8.1                         | 
| Language                            |  C++                         | 
| Segment			|  Robot                        | 

### Issue Description
Hi,
I want to get a specific pixel's RGB value. I don't know how.
Can you give me some advice? Thanks a lot!

dorodnic, 2019-03-04:
Hi @TouchDeeper 
I think something like this:
```
int i = 100, j = 100; // fetch pixel 100,100
rs2::frame rgb = ...
auto ptr = (uint8_t*)rgb.get_data();
auto stride = rgb.as<rs2::video_frame>().stride();
cout << "R=" << ptr[3*(i * stride + j)];
cout << ", G=" << ptr[3*(i * stride + j) + 1];
cout << ", B=" << ptr[3*(i * stride + j) + 2];
```

TouchDeeper, 2019-03-12:
Hi,@dorodnic
Thanks for your advice.

closed 2019-03-12 by TouchDeeper

KlopDesign, 2020-05-21:
Hi, @dorodnic Thanks for the code. I'm now working on it (with a D435), but if I try to get the colorvalues from pixel **(x,y)=1000,1000** I get a pointer read access violation. With the **(x,y)=100,100** it works fine. This is the error: `Exception thrown: read access violation. ptr was 0x11103131EB9192A.`

[image]

I set the `enable_stream` to `cfg.enable_stream(RS2_STREAM_COLOR, WIDTH_COLOR_FRAME, HEIGTH_COLOR_FRAME, RS2_FORMAT_RGB8, 15);` and in my .h file are:
```
#define WIDTH_COLOR_FRAME   1920
#define HEIGTH_COLOR_FRAME  1080
```

This is my code:
```
frameset frames = pl.wait_for_frames();
frame color = frames.get_color_frame();

uint8_t* ptr = (uint8_t*)color.get_data();
int stride = color.as<video_frame>().get_stride_in_bytes();

int i = 1000, j = 1000; // fetch pixel 100,100

cout << "R=" << int(ptr[3 * (i * stride + j)]);
cout << ", G=" << int(ptr[3 * (i * stride + j) + 1]);
cout << ", B=" << int(ptr[3 * (i * stride + j) + 2]);
cout << endl;
```

Maybe you (or someone else of course) can help me? I also asked the question on Stackoverflow btw but didn't get anything at the moment... Thanks in advance!

KlopDesign, 2020-05-25:
Thanks to _nayab_ on StackOverflow, the code should be: 
```
cout << "  R= " << int(ptr[i * stride + (3*j)    ]);
cout << ", G= " << int(ptr[i * stride + (3*j) + 1]);
cout << ", B= " << int(ptr[i * stride + (3*j) + 2]);
```