realsenseai/librealsense #3365: What will cause frame.get_distance(u, v)  to crash?

TouchDeeper, 2019-03-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   |05.11.01.00| 
| Operating System & Version |    Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0                                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.8.1                         | 
| Language                            |  C++                         | 
| Segment			|  Robot                        | 

### Issue Description
Hi,
I want to get depth data of ROI.
``` 
    cout<<"u="<<upixel[0]<<",v="<<upixel[1]<<endl;
    auto udist = frame.get_distance(upixel[0], upixel[1]);
    cout<<"u="<<upixel[0]<<",v="<<upixel[1]<<endl;
```
the output:
```
u=737,v=363
u=737,v=363
u=738,v=363
u=738,v=363
u=739,v=363
u=739,v=363
u=540,v=364
```
As you can see, the code crash when `u=540,v=364,`otherwise，there will have double `u=540,v=364`. And the point which code crash at will change  everytime I run the  code and move the camera.
I guess will it be some invalid data in this depth frame?  If so, how do I deal with this issue?
Can you give me some advice? Thanks a lot!

TouchDeeper, 2019-03-02:
I debug the code,. The GDB output information:
```
Thread 1 "rs-pointcloud" received signal SIGSEGV, Segmentation fault.
0x00007ffff707a026 in librealsense::depth_frame::get_distance (this=0xa5c8a8, x=<optimized out>, y=<optimized out>) at /home/wang/interRealsense/librealsense-2.18.1/src/archive.h:315
315	            case 2: pixel = reinterpret_cast<const uint16_t*>(get_frame_data())[y*get_width() + x]; break;
```
Anyone help?

TouchDeeper, 2019-03-04:
This is not the problem of get_distance(u, v).

closed 2019-03-04 by TouchDeeper

valiafetisov, 2019-10-24:
Hey @TouchDeeper, I'm experiencing the same problem. Calling get_distance with some values crashes the python wrapper with `Segmentation fault`. What was your solution?