realsenseai/librealsense #3369: How to upgrade DFU firmware on Raspbian Linux 2018-11-13 (version)

bittnt, 2019-03-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (current 05.09.02.00 ) | 
| Operating System & Version |   {Rasbian 16.04  | 
| Kernel Version (Linux Only)    |  4.14.98-V7+                                         | 
| Platform                                 | Raspberry Pi 3 B+ |
| SDK Version                            |  { }                          | 
| Language                            |  {C/opencv/python}                          | 
| Segment			|  {others }                          | 

### Issue Description
I follow the exact step in the [documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md). However, after I launch `realsense-viewer`, It tells me that the minimum firmware version is 05.10.03.00.

I went to your DFU update [documentation](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf), it turns out that the aws hosted build would not support arm processor.

Question: Is the documentation `installation_raspbian.md` out of date? How shall i upgrade my firmware? Given that I do not have source code for that to build and you guys did not provide one to support arm processor (which is commonly used in raspberry pi), I with other intel realsense users are very unhappy about this.

dorodnic, 2019-03-04:
At the moment there is no way to update firmware on non-Intel platforms. 
You can most likely do it on your development PC. 
This capability will be provided in future releases.

closed 2019-03-21 by RealSenseCustomerSupport

jgwinner, 2019-03-27:
This is a critical feature.

We are attempting to ship a product based on the Pi; it would be weird if we couldn't update the Realsense cameras in the field. I mean, we can't ask  consumers to plug their cameras into a PC, download some stuff, and do the update.

I realize the Pi has USB 2.0 only, but still ...

I've also found the Intel updates don't work either, but will open a new ticket for that.

Thanks for your consideration,

== John ==

sjdrc, 2019-07-08:
This is also a critical missing feature for us. We have deployed several ARM devices with RealSense cameras to a client in another city. Due to the recent firmware bug that causes some D415s to show up as D410s, our software is not able to run on some of the deployed devices. Only fix is to update the firmware of the device, however I am not able to run the update tool from our remote devices. What am I supposed to do now?

GrubbyHalo, 2019-11-15:
Yes it would be great if this feature could be expedited. We also need to have the firmware update with Raspberry pi's in the field.

GrubbyHalo, 2019-11-22:
I have verified that it is possible to remotely update the firmware on a realsense d435 camera by installing usbip and setting up a usbip service on the pi and connecting remotely to it from a device that can run the firmware updater tool.