realsenseai/librealsense #3370: Additional lights, what wave length?

fiskdata, 2019-03-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 

Hi, 
I have an application with the D435 and I also use a second camera for video capture. We need to illuminate the video with additional IR lights during the night to get a good video. 

I can get this additional illumination with 850 or 940 nm wavelength. Would extra light source of 850nm disturb the Realsense IR-projector and readings more than the 940nm? 

Does anyone have experience with this?

MartyG-RealSense, 2019-03-04:
Intel's illustrated camera tuning guide says in its section about external projectors: "It is possible to use light in visible or near-infrared (ranging from ~400-1000 nm).  Example: 850 nm is invisible to the human eye whilst still being visible to the IR sensor".

This would suggest that either 850 or 940 nm light should be okay.  One of the Intel guys on this forum can confirm that for you though.

The full tuning guide can be viewed at the link below.

https://realsense.intel.com/wp-content/uploads/sites/63/BKM-For-Tuning-D435-and-D415-Cameras-Webinar_Rev3.pdf?language=en_US

fiskdata, 2019-03-04:
Thank you for your reply. 
Maybe the question was not so clear. I was asking if a extra light source of 850nm or 940nm would affect the Realsense camera in different ways or if it didn't matter. The IR projector is emitting at about 850nm, so my first thought would be that a 940nm extra light source would be better. But the other camera used in the system is better at "seeing" 850nm.

agrunnet, 2019-03-04:
The RealSense camera sees both so in that way it does not matter, but there are some subtle aspects to consider. 1. If the scene has a lot of texture (not just flat white walls) then any uniform illumination actually helps the camera. However 2. The RealSense camera is less sensitive to 940nm so if you don’t want the RealSense camera to see the light this might be better to use. 3. You should consider what you want from the other camera - do you want to see the IR projector dots from the RealSense camera? If not then use 940nm and add a long pass filter to that camera.

closed 2019-04-10 by RealSenseCustomerSupport