realsenseai/librealsense #3371: Depth and color image are largely mismatched

jinfagang, 2019-03-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

The combine result shows:
[image]

That depth and color image are not well aligned.

I have already aligned depth to color frame with this code:

```
 rs2::context ctx;
  auto list = ctx.query_devices(); // Get a snapshot of currently connected devices
  cout << "Find " << list.size() << " devices.\n";
  if (list.size() == 0)
    throw std::runtime_error("No device detected. Is it plugged in?");
  rs2::device dev = list.front();
  rs_real_distance_scale = getDepthScale(dev);
  cout << "distance scale: " << rs_real_distance_scale << endl;
  cout << "start pipe ...\n";

  // create a pipe first
  rs2::pipeline pipe;
  rs2::config cfg;
  // config stream, rgb 640x480, 30 fps, as well as depth
  cfg.enable_stream(RS2_STREAM_COLOR, frame_w, frame_h, RS2_FORMAT_BGR8, 30);
  cfg.enable_stream(RS2_STREAM_DEPTH, frame_w, frame_h, RS2_FORMAT_Z16, 30);
  // using for align
  rs2::align align(RS2_STREAM_DEPTH);
  pipe.start(cfg);

  // preload camera
  rs2::frameset frames;
  for (size_t i = 0; i < 30; i++) {
    frames = pipe.wait_for_frames();
    // wating for stream genereated
  }
  cout << "preloading done.\n";

  int c;
  while (c != 27) {

    frames = pipe.wait_for_frames();
    rs2::frameset processed = align.process(frames);

    rs2::frame color_frame = frames.get_color_frame();
    rs2::frame depth_frame = frames.get_depth_frame();
// align depth has no effect
 rs2::frame depth_frame_aligned = processed.first(RS2_STREAM_DEPTH);
    Mat color(Size(frame_w, frame_h), CV_8UC3, (void *) color_frame.get_data(), Mat::AUTO_STEP);
    Mat depth(Size(frame_w, frame_h), CV_16UC1, (void *) depth_frame.get_data(), Mat::AUTO_STEP);
    cv::Mat align_test;
    combine_color_depth(depth, color, align_test);

    imshow("realsense color", color);
    imshow("realsense depth", depth);
    imshow("align test", align_test);
    c = waitKey(1);
  }
```

Try align depth to color does not work well

dorodnic, 2019-03-04:
Hi @jinfagang 
Please check out [align](https://github.com/IntelRealSense/librealsense/tree/master/examples/align) example

ev-mp, 2019-03-04:
@jinfagang , fixing the line
>Mat depth(Size(frame_w, frame_h), CV_16UC1, (void *) ~~depth_frame~~ 
 **depth_frame_aligned**.get_data(), Mat::AUTO_STEP);

should rectify this

RealSenseCustomerSupport, 2019-04-12:
----------------------------------------------

If no further question, will

jinfagang, 2019-04-12:
@ev-mp Already test that, but there still have slightly offset between depth and RGB image, doesn't know if there any effect on final 3D points calculation

closed 2019-04-19 by RealSenseCustomerSupport