realsenseai/librealsense #3372: ubuntu16 realsense D435 SDK installation error

zxfever, 2019-03-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D400 | 
| Firmware Version                   |5.11.0| 
| Operating System & Version |   Win 10| 
| Kernel Version (Linux Only)    |  4.15.0                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.18.1>}                          | 
| Language                            |  C                         | 
| Segment			|  Robot                         | 

### Issue Description
Hello all
I follow the installation tutorial https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md ,when i process the `sudo make uninstall && make clean && make && sudo make install`,  i get the error information as listed follow(I omitted some process without problems.):

[  0%] Building C object third-party/realsense-file/CMakeFiles/realsense-file.dir/lz4/lz4.c.o
[  1%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/bag.cpp.o
[  1%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/buffer.cpp.o
...
[ 23%] Building CXX object CMakeFiles/realsense2.dir/src/l500/l500.cpp.o
[ 23%] Building CXX object CMakeFiles/realsense2.dir/src/l500/l500-private.cpp.o
[ 24%] Building CXX object CMakeFiles/realsense2.dir/src/linux/backend-v4l2.cpp.o
In file included from /home/zxf/librealsense/src/linux/backend-v4l2.cpp:6:0:
/home/zxf/librealsense/src/linux/backend-v4l2.h:54:111: note: #pragma message: 
Librealsense notification: V4L2_META_FMT_UVC was not defined, adding metadata constructs
 e notification: V4L2_META_FMT_UVC was not defined, adding metadata constructs")
                                                                               ^
[ 25%] Building CXX object CMakeFiles/realsense2.dir/src/linux/backend-hid.cpp.o
[ 25%] Building CXX object CMakeFiles/realsense2.dir/src/media/record/record_device.cpp.o
[ 26%] Building CXX object CMakeFiles/realsense2.dir/src/media/record/record_sensor.cpp.o
...
[ 99%] Linking CXX executable rs-rosbag-inspector
[ 99%] Built target rs-rosbag-inspector
Scanning dependencies of target rs-convert
[ 99%] Building CXX object tools/convert/CMakeFiles/rs-convert.dir/rs-convert.cpp.o
[100%] Linking CXX executable rs-convert
please give me some advice to solve this problem,thanks very much!

ev-mp, 2019-03-04:
@zxfever , I cannot see any issue reported in the attached log. Please further elaborate which modules fail to compile/install/run.

dorodnic, 2019-03-04:
Hi @zxfever 
I do not necessarily see a problem, are you sure make fails?

zxfever, 2019-03-04:
Hello @dorodnic 
  There are some instructions below the 24% process, although there is no the 'error' flag, but I don't know if this is normal in the `make` process? After `make` process, I still can't open the realsense D435 (although I updated the firmware). 
  In addition, I once again compiled and found that there is a fatal(I bolded the font) in the process of `cmake ..`,but the final result shows that cmake succeeded. Does this affect the `make `process? I follow somebody's suggestion to apply the` git init` operation to eliminate this fatal, but it did't work. So I don't know what to do. . .The following is my cmake process

-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Info: REALSENSE_VERSION_STRING=2.16.5
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Setting Unix configurations
-- Checking internet connection...
-- Internet connection identified, enabling BUILD_WITH_TM2
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for one of the modules 'libusb-1.0'
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Checking for module 'gtk+-3.0'
--   Found gtk+-3.0, version 3.18.9
-- Building with TM2
-- ----------------------------------------------------------------------------
-- Product versions:
-- - HOST 0.19.3.1175 (Default from versions.cmake)
-- - Remote FW 0.0.18.3452 (Default from versions.cmake)
-- - Remote CENTRAL APP 2.0.19.269 (Default from versions.cmake)
-- - Remote CENTRAL BL 1.0.1.112 (Default from versions.cmake)
-- --------------------------------------------------------------------------------------------------------------------------------------------------------------
-- Downloading FW 0.0.18.3452 from 'http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.0.18.3452/target-0.0.18.3452.mvcmd'
-- Downloading FW 0.0.18.3452 completed - status 0;"No error"
-- Converting FW version 0.0.18.3452 /home/zxf/librealsense-master/third-party/libtm/resources/target.mvcmd to /home/zxf/librealsense-master/third-party/libtm/libtm/src/fw.h
-- Creating HEX buffer from /home/zxf/librealsense-master/third-party/libtm/resources/target.mvcmd offset 0
-- --------------------------------------------------------------------------------------------------------------------------------------------------------------
-- Downloading Central App 2.0.19.269 from 'http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/app/2.0.19.269/central_app-2.0.19.269.bin'
-- Downloading Central App 2.0.19.269 completed - status 0;"No error"
-- Converting Central App version 2.0.19.269 /home/zxf/librealsense-master/third-party/libtm/resources/central_app.bin to /home/zxf/librealsense-master/third-party/libtm/libtm/src/CentralAppFw.h
-- Central Header:
-- - header_size         = 23
-- - data_size           = 42398
-- - file_format_version = 1
-- - version_size        = 7
-- - version             = 2.0.19.269
-- Creating HEX buffer from /home/zxf/librealsense-master/third-party/libtm/resources/central_app.bin offset 23
-- --------------------------------------------------------------------------------------------------------------------------------------------------------------
-- Downloading Central BL 1.0.1.112 from 'http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/bl/1.0.1.112/central_bl-1.0.1.112.bin'
-- Downloading Central BL 1.0.1.112 completed - status 0;"No error"
-- Converting Central BL version 1.0.1.112 /home/zxf/librealsense-master/third-party/libtm/resources/central_bl.bin to /home/zxf/librealsense-master/third-party/libtm/libtm/src/CentralBlFw.h
-- Creating HEX buffer from /home/zxf/librealsense-master/third-party/libtm/resources/central_bl.bin offset 3
**fatal: Not a git repository (or any of the parent directories): .git**
-- --------------------------------------------------------------------------------------------------------------------------------------------------------------
-- Building libtm project on , LIBTM version [0.19.3.1175], API version [10.0], branch [], FW [0.0.18.3452], Central APP [2.0.19.269], Central BL [1.0.1.112]
-- Creating version file /home/zxf/librealsense-master/third-party/libtm/libtm/src/Version.h
-- Building project tm as STATIC library lib
-- --------------------------------------------------------------------------------------------------------------------------------------------------------------
-- Building all projects of libtm_samples
-- Building project libtm_util
-- ----------------------------------------------------------------------------
-- CMake Done
-- ----------------------------------------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zxf/librealsense-master/build

RealSenseCustomerSupport, 2019-03-08:
----------------------------------------------

Hi zefever,

I tried to build from RS 2.18.0 source, and first of all I don't have any error in cmake stage.
After built, using rs-enuerate-device tool, I can detect my D435 with FW 5.11.0.

Would you mind try again? And also make sure that nothing is msised as asked in installation.md?

Below I posted my cmake log, env info, rs-enumerate-device result with D435 FW5.11.1

1. cmake log:
ros@ros-desktop:~/workspace/rs_2.18.1/bld$ cmake .. -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true
-- The CXX compiler identification is GNU 5.5.0
-- The C compiler identification is GNU 5.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Info: REALSENSE_VERSION_STRING=2.18.1
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Setting Unix configurations
-- Checking internet connection...
-- Internet connection identified, enabling BUILD_WITH_TM2
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for one of the modules 'libusb-1.0'
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE
-- Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR)
-- Using X11 for window creation
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so
-- Checking for module 'gtk+-3.0'
--  Found gtk+-3.0, version 3.18.9
-- Building with TM2
-- ----------------------------------------------------------------------------
-- T265 Product versions:
-- - HOST 0.19.3.1505 (Default from versions.cmake)
-- - Remote FW 0.0.18.4577 (Default from versions.cmake)
-- - Remote CENTRAL APP 2.0.19.271 (Default from versions.cmake)
-- - Remote CENTRAL BL 1.0.1.112 (Default from versions.cmake)
-- --------------------------------------------------------------------------------------------------------------------------------------------------------------
-- - Downloading FW 0.0.18.4577 from http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.0.18.4577/target-0.0.18.4577.mvcmd
--  Converting FW version 0.0.18.4577 from target.mvcmd to /home/ros/workspace/rs_2.18.1/third-party/libtm/libtm/src/fw.h
[... 45 lines cut ...]
ros@ros-desktop:~/workspace/rs_2.18.1/bld$ uname -a
Linux ros-desktop 4.15.0-45-generic #48~16.04.1-Ubuntu SMP Tue Jan 29 18:03:48 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
ros@ros-desktop:~/workspace/rs_2.18.1/bld$
ros@ros-desktop:~/workspace/rs_2.18.1/bld$
ros@ros-desktop:~/workspace/rs_2.18.1/bld$ cmake --version
cmake version 3.11.4
CMake suite maintained and supported by Kitware (kitware.com/cmake).
ros@ros-desktop:~/workspace/rs_2.18.1/bld$
ros@ros-desktop:~/workspace/rs_2.18.1/bld$
ros@ros-desktop:~/workspace/rs_2.18.1/bld$ gcc --version
gcc (Ubuntu 5.5.0-12ubuntu1~16.04) 5.5.0 20171010
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

4. rs-enurate-device result

ros@ros-desktop:~/workspace/rs_2.18.1/bld$
ros@ros-desktop:~/workspace/rs_2.18.1/bld$ cd tools/enumerate-devices/
ros@ros-desktop:~/workspace/rs_2.18.1/bld/tools/enumerate-devices$ ./rs-enumerate-devices -s
Device Name                  Serial Number      Firmware Version
Intel RealSense D435          728612070408        05.11.01.00
ros@ros-desktop:~/workspace/rs_2.18.1/bld/tools/enumerate-devices$

Thanks!

ev-mp, 2019-03-11:
@zxfever,
The reported error indicates git-related issue/repo corruption:
>fatal: Not a git repository (or any of the parent directories): .git

 - Are you downloading the sources as zip/tar.gz or via `git clone` ? [Make sure](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#ubuntu-build-dependencies) that `git` is properly installed.
 - Delete all files (including root-owned) in librealsense/build and rerun cmake build.
 - Make a fresh download into a new location and rebuild

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Hi zefever,

Do you have any update on this one? Or do you still need help?

Thanks!

closed 2019-04-02 by RealSenseCustomerSupport