realsenseai/librealsense #3373: Debian / Ubuntu packages building

jgomezlopez, 2019-03-04:
I'm trying to build packages for Ubuntu and Debian using catkin as I saw you have a package.xml file. The main aim is to build packages for ARM. I'm trying first in a X86_64 platform. If I run 

catkin build --cmake-args -DBUILD_GRAPHICAL_EXAMPLES=false -DBUILD_UNIT_TESTS=false -DBUILD_EASYLOGGINGPP=false

I get the following output

```
Profile:                     default
Extending:                   None
Workspace:                   /librealsense
------------------------------------------------------------------------------------------------------------------
Source Space:       [exists] /librealsense/src
Log Space:         [missing] /librealsense/logs
Build Space:        [exists] /librealsense/build
Devel Space:        [exists] /librealsense/devel
Install Space:      [unused] /librealsense/install
DESTDIR:            [unused] None
------------------------------------------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
------------------------------------------------------------------------------------------------------------------
Additional CMake Args:       -DBUILD_GRAPHICAL_EXAMPLES=false -DBUILD_UNIT_TESTS=false -DBUILD_EASYLOGGINGPP=false
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
------------------------------------------------------------------------------------------------------------------
NOTE: Forcing CMake to run for each package.
------------------------------------------------------------------------------------------------------------------

------------------------------------------------------------------------------------------------------------------
WARNING: Your workspace is not extending any other result space, but it is set to use a `linked` devel space
layout. This requires the `catkin` CMake package in your source space in order to be built.
------------------------------------------------------------------------------------------------------------------

[build] No packages were found in the source space '/librealsense/src'
[build] No packages to be built.
[build] Package table is up to date.                                                                                                
Starting  >>> catkin_tools_prebuild                                                                                                 
____________________________________________________________________________________________________________________________________
Errors     << catkin_tools_prebuild:cmake /librealsense/logs/catkin_tools_prebuild/build.cmake.000.log                              
CMake Error at /librealsense/build/catkin_tools_prebuild/CMakeLists.txt:12 (message):
  The catkin CMake module was not found, but it is required to build a linked
  workspace.  To resolve this, please do one of the following, and try
  building again.

   1. Source the setup.sh file from an existing catkin workspace:
      source SETUP_FILE

   2. Extend another catkin workspace's result (install or devel) space:
      catkin config --extend RESULT_SPACE

   3. Set `catkin_DIR` to the directory containing `catkin-config.cmake`:
      catkin config --cmake-args -Dcatkin_DIR=CATKIN_CMAKE_CONFIG_PATH

   4. Add the catkin source package to your workspace's source space:
      cd SOURCE_SPACE && git clone https://github.com/ros/catkin.git

cd /librealsense/build/catkin_tools_prebuild; catkin build --get-env catkin_tools_prebuild | catkin env -si  /usr/bin/cmake /librealsense/build/catkin_tools_prebuild --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/librealsense/devel/.private/catkin_tools_prebuild -DCMAKE_INSTALL_PREFIX=/librealsense/install -DBUILD_GRAPHICAL_EXAMPLES=false -DBUILD_UNIT_TESTS=false -DBUILD_EASYLOGGINGPP=false; cd -
....................................................................................................................................
Failed     << catkin_tools_prebuild:cmake          [ Exited with code 1 ]                                                           
Failed    <<< catkin_tools_prebuild                [ 26.6 seconds ]                                                                 
[build] Summary: 0 of 1 packages succeeded.                                                                                         
[build]   Ignored:   None.                                                                                                          
[build]   Warnings:  None.                                                                                                          
[build]   Abandoned: None.                                                                                                          
[build]   Failed:    1 packages failed.                                                                                             
[build] Runtime: 26.8 seconds total.                                                                                                
Exception ignored in: <bound method BaseEventLoop.__del__ of <_UnixSelectorEventLoop running=False closed=True debug=False>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/trollius/base_events.py", line 395, in __del__
  File "/usr/lib/python3/dist-packages/trollius/unix_events.py", line 65, in close
  File "/usr/lib/python3/dist-packages/trollius/unix_events.py", line 166, in remove_signal_handler
  File "/usr/lib/python3.5/signal.py", line 47, in signal
TypeError: signal handler must be signal.SIG_IGN, signal.SIG_DFL, or a callable object
```
What commands sequence I need to execute?
 
Thanks in advance for your help

abylikhsanov, 2019-03-04:
Well, it is written exactly in your log, you do not have CMake which is essential for compilation and linking. Make sure you have CMake on your machine.

jgomezlopez, 2019-03-05:
Cmake 3.5 is installed, the minimum version required is 3.1. But even if I try to compile the sources using
 `cmake ../ -DBUILD_GRAPHICAL_EXAMPLES=false -DBUILD_UNIT_TESTS=false -DBUILD_WITH_OPENMP=false -DCMAKE_BUILD_TYPE=Release -DBUILD_EASYLOGGINGPP=false`
and
`make`
I get 
`[ 27%] Building CXX object CMakeFiles/realsense2.dir/src/linux/backend-hid.cpp.o
In file included from /librealsense/src/linux/backend-hid.cpp:6:0:
/librealsense/src/linux/backend-hid.h: In function 'bool librealsense::platform::write_fs_attribute(const string&, const T&)':
/librealsense/src/linux/backend-hid.h:46:27: error: invalid use of incomplete type 'std::fstream {aka class std::basic_fstream<char>}'
             if (!fs_handle.good())
                           ^
In file included from /usr/include/c++/5/memory:72:0,
                 from /librealsense/src/backend.h:11,
                 from /librealsense/src/linux/backend-hid.h:5,
                 from /librealsense/src/linux/backend-hid.cpp:6:
/usr/include/c++/5/iosfwd:122:11: note: declaration of 'std::fstream {aka class std::basic_fstream<char>}'
     class basic_fstream;
           ^
In file included from /librealsense/src/linux/backend-hid.cpp:6:0:
/librealsense/src/linux/backend-hid.h:59:30: error: invalid use of incomplete type 'std::fstream {aka class std::basic_fstream<char>}'
                     fs_handle.close();
                              ^
In file included from /usr/include/c++/5/memory:72:0,
                 from /librealsense/src/backend.h:11,
                 from /librealsense/src/linux/backend-hid.h:5,
                 from /librealsense/src/linux/backend-hid.cpp:6:
/usr/include/c++/5/iosfwd:122:11: note: declaration of 'std::fstream {aka class std::basic_fstream<char>}'
     class basic_fstream;
           ^
In file included from /librealsense/src/linux/backend-hid.cpp:6:0:
/librealsense/src/linux/backend-hid.h:60:30: error: invalid use of incomplete type 'std::fstream {aka class std::basic_fstream<char>}'
                     fs_handle.open(path);
                              ^
In file included from /usr/include/c++/5/memory:72:0,
                 from /librealsense/src/backend.h:11,
                 from /librealsense/src/linux/backend-hid.h:5,
                 from /librealsense/src/linux/backend-hid.cpp:6:
/usr/include/c++/5/iosfwd:122:11: note: declaration of 'std::fstream {aka class std::basic_fstream<char>}'
     class basic_fstream;
           ^
In file included from /librealsense/src/linux/backend-hid.cpp:6:0:
/librealsense/src/linux/backend-hid.h:62:30: error: invalid use of incomplete type 'std::fstream {aka class std::basic_fstream<char>}'
                     fs_handle.flush();
                              ^
In file included from /usr/include/c++/5/memory:72:0,
                 from /librealsense/src/backend.h:11,
                 from /librealsense/src/linux/backend-hid.h:5,
                 from /librealsense/src/linux/backend-hid.cpp:6:
/usr/include/c++/5/iosfwd:122:11: note: declaration of 'std::fstream {aka class std::basic_fstream<char>}'
     class basic_fstream;
           ^
/librealsense/src/linux/backend-hid.cpp: In member function 'void librealsense::platform::hid_input::enable(bool)':
/librealsense/src/linux/backend-hid.cpp:59:55: error: variable 'std::ofstream iio_device_file' has initializer but incomplete type
             std::ofstream iio_device_file(element_path);
                                                       ^
/librealsense/src/linux/backend-hid.cpp: In member function 'void librealsense::platform::hid_input::init()':
/librealsense/src/linux/backend-hid.cpp:84:63: error: variable 'std::ifstream device_type_file' has initializer but incomplete type
             std::ifstream device_type_file(read_scan_type_path);
                                                               ^
/librealsense/src/linux/backend-hid.cpp:121:65: error: variable 'std::ifstream device_index_file' has initializer but incomplete type
             std::ifstream device_index_file(read_scan_index_path);
                                                                 ^
/librealsense/src/linux/backend-hid.cpp:134:69: error: variable 'std::ifstream device_enabled_file' has initializer but incomplete type
             std::ifstream device_enabled_file(read_enable_state_path);
                                                                     ^
/librealsense/src/linux/backend-hid.cpp: In member function 'void librealsense::platform::hid_custom_sensor::start_capture(librealsense::platform::hid_callback)':
/librealsense/src/linux/backend-hid.cpp:202:59: error: variable 'std::ifstream device_file' has initializer but incomplete type
             std::ifstream device_file(read_device_path_str);
                                                           ^
/librealsense/src/linux/backend-hid.cpp: In member function 'void librealsense::platform::hid_custom_sensor::enable(bool)':
[... 93 lines cut ...]
                     std::ifstream vnv_handle(path);
                                                  ^
/librealsense/src/linux/backend-hid.h: In instantiation of 'bool librealsense::platform::write_fs_attribute(const string&, const T&) [with T = std::__cxx11::basic_string<char>; std::__cxx11::string = std::__cxx11::basic_string<char>]':
/librealsense/src/linux/backend-hid.cpp:746:69:   required from here
/librealsense/src/linux/backend-hid.h:45:40: error: 'std::fstream fs_handle' has incomplete type
[... cut ...]
                     std::ifstream vnv_handle(path);
                                                  ^
CMakeFiles/realsense2.dir/build.make:470: recipe for target 'CMakeFiles/realsense2.dir/src/linux/backend-hid.cpp.o' failed
make[2]: *** [CMakeFiles/realsense2.dir/src/linux/backend-hid.cpp.o] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/realsense2.dir/all' failed
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
`

abylikhsanov, 2019-03-05:
Did you try to use `cmake ../` without any options?

jgomezlopez, 2019-03-05:
I didn't try, but compilation works with 2.18.0, I will try with 2.18.1 and I will send you the logs

jgomezlopez, 2019-03-05:
I can't run cmake without options since I haven't installed the graphical environment (Ubuntu server)
The cmake output is:
`-- Info: REALSENSE_VERSION_STRING=2.18.1
-- Setting Unix configurations
-- Checking internet connection...
-- Internet connection identified, enabling BUILD_WITH_TM2
-- Building with TM2
-- ----------------------------------------------------------------------------
-- T265 Product versions:
-- - HOST 0.19.3.1505 (Default from versions.cmake)
-- - Remote FW 0.0.18.4577 (Default from versions.cmake)
-- - Remote CENTRAL APP 2.0.19.271 (Default from versions.cmake)
-- - Remote CENTRAL BL 1.0.1.112 (Default from versions.cmake)
-- --------------------------------------------------------------------------------------------------------------------------------------------------------------
-- - Downloading FW 0.0.18.4577 from http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.0.18.4577/target-0.0.18.4577.mvcmd
--   Converting FW version 0.0.18.4577 from target.mvcmd to /librealsense/third-party/libtm/libtm/src/fw.h
--   File /librealsense/third-party/libtm/libtm/src/fw.h wasn't changed
-- - Downloading Central App 2.0.19.271 from http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/app/2.0.19.271/central_app-2.0.19.271.bin
--   Converting Central App version 2.0.19.271 from central_app.bin to /librealsense/third-party/libtm/libtm/src/CentralAppFw.h
--   File /librealsense/third-party/libtm/libtm/src/CentralAppFw.h wasn't changed
-- - Downloading Central BL 1.0.1.112 from http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/bl/1.0.1.112/central_bl-1.0.1.112.bin
--   Converting Central BL version 1.0.1.112 from central_bl.bin to /librealsense/third-party/libtm/libtm/src/CentralBlFw.h
--   File /librealsense/third-party/libtm/libtm/src/CentralBlFw.h wasn't changed
-- --------------------------------------------------------------------------------------------------------------------------------------------------------------
-- Building libtm project on , LIBTM version [0.19.3.1505], API version [10.0], branch [master], FW [0.0.18.4577], Central APP [2.0.19.271], Central BL [1.0.1.112]
-- Creating version file /librealsense/third-party/libtm/libtm/src/Version.h
-- Building project tm as STATIC library lib
-- --------------------------------------------------------------------------------------------------------------------------------------------------------------
-- Building all projects of libtm_samples
-- Building project libtm_util
-- ----------------------------------------------------------------------------
-- CMake Done
-- ----------------------------------------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /librealsense/build
`
I haven´t found any log file

jgomezlopez, 2019-03-05:
I closed it accidentally

130s, 2019-03-12:
`librealsense` is a `cmake` package (see this https://github.com/IntelRealSense/librealsense/pull/1602, or [this line](https://github.com/IntelRealSense/librealsense/pull/1602/files#diff-34ae68f4adad56c25c5bc05dcb64794eR36) that made the change). So `catkin_make_install` is the best tool from catkin family to build.

I'm not yet so sure how well catkin_tools supports building a pure `cmake` package.

jgomezlopez, 2019-03-13:
Thanks for the suggestion. What I'm doing is generating the Debian packages using debhelp. It is mostly finished, so I can do a PR to include those files or create a separate project.

130s, 2019-03-18:
FYI I opened a request to the maintainers to restart making binary releases https://github.com/IntelRealSense/librealsense/issues/3528

jgomezlopez, 2019-03-19:
Thanks for doing that. I finished the packaging in the following way. I created librealsense2, librealsense2-dev and rs2-utils packages. I'm thinking in adding the udev rules file in another package and add a dependency in librealsense package. I don't know how to handle the kernel patching, maybe creating a new package and create a suggested package entry in the librealsense.
First I will create a new package for the udev rules file and send a PR to the mantainers.

closed 2024-03-29 by MartyG-RealSense