realsenseai/librealsense #3374: D435i stops sending frames after a minute

abylikhsanov, 2019-03-04:
| Camera Model                       | D435i| 
| Firmware Version                   |5.10.13.0| 
| Operating System & Version |   Ubuntu 18.04 ARM | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Odroid XU4 |
| SDK Version                            |  { legacy / 2.18.1}                          | 

### Issue Description

I have installed SDK from sourcce and patched the kernel for Odroid. After that, I was able to detect the device using USB 3.0

When I open the realsense-viewer, I can see that the camera is being detected but after a minute I get no frames and the error I get is:

```
04/03 13:06:48,574 WARNING [2821296624] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
 04/03 13:07:08,465 ERROR [2804511216] (types.h:214) map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 04/03 13:07:08,470 ERROR [2804511216] (error-handling.cpp:66) Error during polling error handler: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 04/03 13:07:13,652 WARNING [2764034544] (backend-hid.h:48) write_fs_attribute with accel_3d-dev1 failed. The specified path /sys/devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.1/4-1.1:1.5/0003:8086:0B3A.0001/HID-SENSOR-200073.8.auto/iio:device1/trigger/current_trigger is not valid
 04/03 13:07:13,654 ERROR [3019571216] (types.h:214) Failed to open scan_element /sys/devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.1/4-1.1:1.5/0003:8086:0B3A.0001/HID-SENSOR-200073.8.auto/iio:device1/scan_elements/in_accel_z_en Last Error: Permission denied
 04/03 13:07:13,655 ERROR [3019571216] (backend-hid.cpp:950) Hid device is busy!
 04/03 13:07:13,660 WARNING [3019571216] (rs.cpp:273) null pointer passed for argument "device"
 04/03 13:07:34,027 WARNING [3019571216] (types.cpp:57) hwmon command 0x4f failed. Error type: No data to return (-21).
 04/03 13:07:34,172 WARNING [3019571216] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
 04/03 13:08:06,644 WARNING [2747249136] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 04/03 13:08:07,862 WARNING [2793398768] (types.cpp:57) get_xu(id=7) failed! Last Error: Input/output error
 04/03 13:08:07,862 ERROR [2793398768] (error-handling.cpp:66) Error during polling error handler: get_xu(id=7) failed! Last Error: Input/output error
 04/03 13:08:11,649 WARNING [2747249136] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 04/03 13:08:14,573 WARNING [2793398768] (types.cpp:57) get_xu(id=7) failed! Last Error: Input/output error
 04/03 13:08:14,573 ERROR [2793398768] (error-handling.cpp:66) Error during polling error handler: get_xu(id=7) failed! Last Error: Input/output error
 04/03 13:08:16,652 WARNING [2747249136] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 04/03 13:08:21,296 WARNING [2793398768] (types.cpp:57) get_xu(id=7) failed! Last Error: Input/output error
 04/03 13:08:21,296 ERROR [2793398768] (error-handling.cpp:66) Error during polling error handler: get_xu(id=7) failed! Last Error: Input/output error
 04/03 13:08:21,653 WARNING [2747249136] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 04/03 13:08:26,656 WARNING [2747249136] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 04/03 13:08:28,052 WARNING [2793398768] (types.cpp:57) get_xu(id=7) failed! Last Error: Input/output error
 04/03 13:08:28,052 ERROR [2793398768] (error-handling.cpp:66) Error during polling error handler: get_xu(id=7) failed! Last Error: Input/output error

```

Here is the dmesg output:

```
uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

 uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
```

ev-mp, 2019-03-04:
@abylikhsanov , please provide the full dmesg log between device connected and frames stop to arrive.

abylikhsanov, 2019-03-05:
@ev-mp Here is the dmesg of failure:

```
[  101.346271] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  103.140511] hid-sensor-hub 0003:8086:0B3A.0001: No report with id 0xffffffff found
[  111.040061] uvcvideo: Non-zero status (-75) in video completion handler.
[  111.040139] uvcvideo: Non-zero status (-71) in video completion handler.
[  111.040392] uvcvideo: Non-zero status (-71) in video completion handler.
[  111.040635] uvcvideo: Non-zero status (-71) in video completion handler.
[  111.040755] uvcvideo: Non-zero status (-71) in video completion handler.
[  111.548802] uvcvideo: Failed to query (GET_CUR) UVC control 7 on unit 3: -71 (exp. 1).
[  111.605783] uvcvideo: Failed to query (GET_CUR) UVC control 7 on unit 3: -71
```

Then I replug the realsense:

```
[  195.573591] usb 4-1.1: new SuperSpeed USB device number 4 using xhci-hcd
[  195.594377] usb 4-1.1: New USB device found, idVendor=8086, idProduct=0b3a
[  195.594386] usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  195.594394] usb 4-1.1: Product: Intel(R) RealSense(TM) Depth Camera 435i
[  195.594402] usb 4-1.1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[  195.594409] usb 4-1.1: SerialNumber: 852323053439
[  195.599098] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  195.600112] uvcvideo: Unable to create debugfs 4-4 directory.
[  195.600331] uvcvideo 4-1.1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  195.600343] uvcvideo 4-1.1:1.0: Entity type for entity Processing 2 was not initialized!
[  195.600354] uvcvideo 4-1.1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  195.600362] uvcvideo 4-1.1:1.0: Entity type for entity Camera 1 was not initialized!
[  195.600762] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.1/4-1.1:1.0/input/input2
[  195.602052] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  195.602517] uvcvideo: Unable to create debugfs 4-4 directory.
[  195.602738] uvcvideo 4-1.1:1.3: Entity type for entity Processing 7 was not initialized!
[  195.602746] uvcvideo 4-1.1:1.3: Entity type for entity Extension 8 was not initialized!
[  195.602754] uvcvideo 4-1.1:1.3: Entity type for entity Camera 6 was not initialized!
[  195.665531] hid-sensor-hub 0003:8086:0B3A.0002: No report with id 0xffffffff found
[  195.666680] hid-sensor-hub 0003:8086:0B3A.0002: No report with id 0xffffffff found
[  195.725515] hid-sensor-hub 0003:8086:0B3A.0002: No report with id 0xffffffff found
[  195.726509] hid-sensor-hub 0003:8086:0B3A.0002: No report with id 0xffffffff found
[  196.193538] hid-sensor-hub 0003:8086:0B3A.0002: No report with id 0xffffffff found
[  198.397585] ------------[ cut here ]------------
[  198.397606] WARNING: CPU: 5 PID: 2850 at fs/sysfs/group.c:237 sysfs_remove_group+0x94/0x9c
[  198.397613] sysfs group 'power' not found for kobject 'event1'
[  198.397619] Modules linked in: fuse cpufreq_conservative cpufreq_userspace cpufreq_powersave loop binfmt_misc hid_sensor_accel_3d hid_sensor_gyro_3d hid_sensor_custom hid_sensor_trigger hid_sensor_iio_common hid_sensor_hub spidev evdev usbhid uvcvideo cdc_acm videobuf2_vmalloc rtl8812au cfg80211 rfkill spi_s3c64xx uio_pdrv_genirq uio gpio_keys extcon_usb_gpio exynos_gpiomem sch_fq_codel ip_tables ipv6
[  198.397799] CPU: 5 PID: 2850 Comm: nodelet Not tainted 4.14.102+ #1
[  198.397804] Hardware name: ODROID-XU4
[  198.397824] [<c0110778>] (unwind_backtrace) from [<c010c7f0>] (show_stack+0x10/0x14)
[  198.397836] [<c010c7f0>] (show_stack) from [<c08ad5d4>] (dump_stack+0x88/0x9c)
[  198.397847] [<c08ad5d4>] (dump_stack) from [<c0124044>] (__warn+0xec/0x104)
[  198.397857] [<c0124044>] (__warn) from [<c0124094>] (warn_slowpath_fmt+0x38/0x48)
[  198.397868] [<c0124094>] (warn_slowpath_fmt) from [<c0301b5c>] (sysfs_remove_group+0x94/0x9c)
[  198.397882] [<c0301b5c>] (sysfs_remove_group) from [<c059e74c>] (device_del+0x48/0x358)
[  198.397895] [<c059e74c>] (device_del) from [<c028ae88>] (cdev_device_del+0x14/0x2c)
[  198.397910] [<c028ae88>] (cdev_device_del) from [<bf3a8178>] (evdev_disconnect+0x1c/0x44 [evdev])
[  198.397934] [<bf3a8178>] (evdev_disconnect [evdev]) from [<c0667074>] (__input_unregister_device+0x9c/0x148)
[  198.397947] [<c0667074>] (__input_unregister_device) from [<c0667844>] (input_unregister_device+0x48/0x70)
[  198.397964] [<c0667844>] (input_unregister_device) from [<bf35e29c>] (uvc_delete+0x14/0x14c [uvcvideo])
[  198.397981] [<bf35e29c>] (uvc_delete [uvcvideo]) from [<c059d598>] (device_release+0x2c/0x90)
[  198.397992] [<c059d598>] (device_release) from [<c08b1500>] (kobject_put+0xe0/0x218)
[  198.398005] [<c08b1500>] (kobject_put) from [<c069aaa4>] (v4l2_release+0x4c/0x78)
[  198.398015] [<c069aaa4>] (v4l2_release) from [<c0288230>] (__fput+0x84/0x1cc)
[  198.398026] [<c0288230>] (__fput) from [<c0141f2c>] (task_work_run+0x90/0xb0)
[  198.398037] [<c0141f2c>] (task_work_run) from [<c0129068>] (do_exit+0x3bc/0xb78)
[  198.398047] [<c0129068>] (do_exit) from [<c012988c>] (do_group_exit+0x3c/0xbc)
[  198.398057] [<c012988c>] (do_group_exit) from [<c0134054>] (get_signal+0x108/0x8c8)
[  198.398068] [<c0134054>] (get_signal) from [<c010baf0>] (do_signal+0xc4/0x450)
[  198.398079] [<c010baf0>] (do_signal) from [<c010c03c>] (do_work_pending+0xb0/0xc8)
[... cut ...]
[  311.812336] hid-sensor-hub 0003:8086:0B3A.0003: No report with id 0xffffffff found
[  311.813503] hid-sensor-hub 0003:8086:0B3A.0003: No report with id 0xffffffff found
[  311.872321] hid-sensor-hub 0003:8086:0B3A.0003: No report with id 0xffffffff found
[  311.873328] hid-sensor-hub 0003:8086:0B3A.0003: No report with id 0xffffffff found
[  312.288382] hid-sensor-hub 0003:8086:0B3A.0003: No report with id 0xffffffff found

```

And that just happens randomely, I cannot say when the camera will work and when not. Most of the time it does not.

ev-mp, 2019-03-05:
@abylikhsanov ,
I cannot point to something specific, but based on the topic description and some hints in kernel log, I'd recommend to try to connect the device to Odroid via a self-powered USB3 hub to rule out power-related issues.

abylikhsanov, 2019-03-05:
@ev-mp I can buy one but most of the hubs can deliver up to 2 amps. Is that enough to power D435i?

ev-mp, 2019-03-05:
D435i power requirements are approximately 3.5W (5V*0.7A). USB3 max permitted is 0.9A.
Most USB3 self-powered hubs should qualify, though a decent quality one would be recommended.

abylikhsanov, 2019-03-05:
Okay I can confirm that this was due to power lack. With USB 3.0 Hub (5v 2a) I did not have any issues.

closed 2019-03-05 by abylikhsanov