realsenseai/librealsense #3375: pcl viewer will conflict with the api frame.get_profile().as<rs2::video_stream_profile>().get_intrinsics()

TouchDeeper, 2019-03-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   |05.11.01.00| 
| Operating System & Version |    Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0                                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.8.1                         | 
| Language                            |  C++                         | 
| Segment			|  Robot                        | 

### Issue Description
Hi,
I seem to have found a bug.
When you put a code like below in the loop of pcl viewer, within the window will be all white and can't show the point properly.
``` 
rs2_intrinsics intr = frame.get_profile().as<rs2::video_stream_profile>().get_intrinsics()
```
Just like the image below
[image]

dorodnic, 2019-03-05:
Hi @TouchDeeper 
Thanks for the report. This is very strange indeed. I'd like our support team to reproduce and we will follow-up accordingly

RealSenseCustomerSupport, 2019-03-11:
----------------------------------------------

Hi @TouchDeeper,

The `get_profile()`API and the pcl point cloud generation seem to work as expected on my side.

A very slight modification from the sample code rs-pcl - https://github.com/IntelRealSense/librealsense/blob/master/wrappers/pcl/pcl/rs-pcl.cpp

      // Wait for the next set of frames from the camera
      auto frames = pipe.wait_for_frames();

      //get sensor intrins
      for (auto&& frame : frames) {

        rs2_intrinsics intrinsics = frame.get_profile().as<rs2::video_stream_profile>().get_intrinsics();

        auto principal_point = std::make_pair(intrinsics.ppx, intrinsics.ppy);
        auto focal_length = std::make_pair(intrinsics.fx, intrinsics.fy);
      }

      auto depth = frames.get_depth_frame();

      // Generate the pointcloud and texture mappings
      points = pc.calculate(depth);

      auto pcl_points = points_to_pcl(points);

RealSense SDK: 2.19.0
PCL: 1.9.1
D435 FW: 5.11.1.100
Host: Ubuntu 16.04_4.4.0-142

Please let me know if there is anything I've missed. Thank you.

Attachment(s):
pcl_get_intrin.png - https://realsensesupport.intel.com/attachments/token/r9x7MorkTQXvZEkiY5MtssoKH/?name=pcl_get_intrin.png

TouchDeeper, 2019-03-12:
Hi,@RealSense Customer Support
Thanks for your test.
I think it is the PCL viewer that conflict with the `rs2_intrinsics intr = frame.get_profile().as<rs2::video_stream_profile>().get_intrinsics()`.
You don't seem to use the PCL viewer.

RealSenseCustomerSupport, 2019-03-12:
----------------------------------------------

Hi @TouchDeeper,

Thank you so much for pointing out the attachment issue.
I made an assumption that the PCL viewer referring to here is the pcl viewer in the PCL library and made something like below to visualize the point cloud with PCLVisualizer in attempt to reproduce your issue.

        //====================
        // Object Declaration
        //====================
        boost::shared_ptr<pcl::visualization::PCLVisualizer> viewer (new pcl::visualization::PCLVisualizer ("Captured Frame"));
        // Declare pointcloud object, for calculating pointclouds and texture mappings
        rs2::pointcloud pc;

        // Begin Stream with default configs
        // Set background of viewer to black
        viewer->setBackgroundColor (0, 0, 0);
        // Default size for rendered points
        viewer->setPointCloudRenderingProperties (pcl::visualization::PCL_VISUALIZER_POINT_SIZE, 1, "Cloud");
        // Viewer Properties
        viewer->initCameraParameters();  // Camera Parameters for ease of viewing
        // Loop and take frame captures upon user input
        while(1) {
            // Capture a single frame and obtain depth + RGB values from it
            auto frames = pipe.wait_for_frames();
            auto depth = frames.get_depth_frame();
            auto RGB = frames.get_color_frame();
            for (auto&& frame : frames) {
                rs2_intrinsics intrinsics = frame.get_profile().as<rs2::video_stream_profile>().get_intrinsics();
                auto principal_point = std::make_pair(intrinsics.ppx, intrinsics.ppy);
                auto focal_length = std::make_pair(intrinsics.fx, intrinsics.fy);
                std::cout << "Principal Point : " << principal_point.first << ", " << principal_point.second << std::endl;
                std::cout << "Focal Length : " << focal_length.first << ", " << focal_length.second << std::endl;
            }
            // Map Color texture to each point
            pc.map_to(RGB);
            // Generate Point Cloud
            auto points = pc.calculate(depth);
            // Convert generated Point Cloud to PCL Formatting
            cloud_pointer cloud = PCL_Conversion(points, RGB);
            viewer->addPointCloud<pcl::PointXYZRGB> (cloud, "Cloud");
            viewer->spinOnce(100); // Allow user to rotate point cloud and view it
            viewer->removePointCloud("Cloud");
        }//End-while

Some clarification and information to help us to reproduce would be highly appreciated. Thank you.

TouchDeeper, 2019-03-13:
Hi, @RealSense Customer Support
I'm sorry to say I have modified the previous code.
Maybe there was another problem cause the problem.
I test your code. It runs well. And your code has given me some tip to simplify my code.
Thanks a lot. Next time I report a bug issue, I will post the code or keep the code unchanged.

RealSenseCustomerSupport, 2019-03-13:
----------------------------------------------

Hi @TouchDeeper,

No worries, and I am glad the code was helpful.
Let me close this issue now.

closed 2019-03-14 by TouchDeeper