realsenseai/librealsense #3376: Different Resolution for depth and color stream to make camera working

JRauer, 2019-03-04:
Hello,
I noticed in the Python-Wrapper Alignment-Example (librealsense/wrappers/python/examples/align-depth2color.py) that different resolutions are applied for the color and the depth stream. I had to make both streams the same resolution in order to make the script work (otherwise it just would not stream anything). Then I updated the camera's firmware to the newest version and now it just streams something, if the resolutions are different (as in the example). Could you explain this behavior? Why are there different resolutions in the example-file?
Best regards in advance!

dorodnic, 2019-03-04:
Hi @JRauer 
Perhaps this is due to the camera being recognized as a USB2 device? Working via USB2 severely limits supported resolutions, and newer firmware unlocks more USB2 resolutions, so this might be the cause. 
There is no strict dependency between RGB and Depth resolutions in general. Please note that Depth and IR must be equal (because depth is generated from IR)

JRauer, 2019-03-05:
Hi @dorodnic ,
thank you for your fast reply! Realsense-viewer shows a USB-3.2 connection so I don't think that this causes the problem. I played around with different resolutions and if I change to 1280x720 and noticed that it works perfectly! I tried in the realsense-viewer changing to 640x480 for the depth camera and wasn't able to get a stream. So somehow D435-Depth module does not support this resolution (anymore). So changing to a different resolution solved the problem for me. Maybe that should be added to the documentation or reported as bug?

dorodnic, 2019-03-05:
We still very much support 640x480 resolution, that's very strange. 
Did you have an opportunity to test the unit on a different computer?
All I can think of is, maybe, calibration table corruption, but even then you would get (black) frames.

JRauer, 2019-03-05:
I'm afraid I don't have another system/camera to test the device but I can send you firmware/software versions if you want to replicate the bug?
I don't beliefe that the calibration table is corrupted. I can read it out and all values do make sense and are equal to before the firmware-update. I don't really have any idea what's causing this problem and if it's a device problem or a software problem.

RealSenseCustomerSupport, 2019-03-11:
----------------------------------------------

@JRauer What's firmware/software versions when you get issue for 640x480?

JRauer, 2019-03-14:
@RealSenseCustomerSupport 
Librealsense 2.18.1 Release
OS Linux 16.04, 64 bit
Graphics Card GeForce GTX 1050Ti
Firmware-version of d435: 5.11.01.100
USB-Type 3.2

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

@JRauer Could you please provide the result of rs-enumerate(https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices )?

JRauer, 2019-03-21:
@RealSenseCustomerSupport  
Device info: 
    Name: 	Intel RealSense D435
    Serial Number: 	819312072632
    Firmware Version: 	05.11.01.100
    Recommended Firmware Version: 	05.10.03.00
    Physical Port: 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0
    Debug Op Code: 	15
    Advanced Mode: 	YES
    Product Id: 	0B07
    Usb Type Descriptor : 	3.2

Stream Profiles supported by Stereo Module
 Supported modes:                                      stream     resolution fps   format   
    Infrared 2	  1280x800	@ 30Hz	Y8
    Infrared 1	  1280x800	@ 30Hz	Y8
    Infrared 1	  1280x800	@ 25Hz	Y16
    Infrared 2	  1280x800	@ 25Hz	Y16
    Infrared 2	  1280x800	@ 15Hz	Y16
    Infrared 1	  1280x800	@ 15Hz	Y16
    Infrared 1	  1280x800	@ 15Hz	Y8
    Infrared 2	  1280x800	@ 15Hz	Y8
    Infrared 1	  1280x720	@ 30Hz	Y8
    Infrared 2	  1280x720	@ 30Hz	Y8
    Infrared 2	  1280x720	@ 15Hz	Y8
    Infrared 1	  1280x720	@ 15Hz	Y8
    Infrared 1	  1280x720	@ 6Hz	Y8
    Infrared 2	  1280x720	@ 6Hz	Y8
    Infrared 2	  848x480	@ 90Hz	Y8
    Infrared 1	  848x480	@ 90Hz	Y8
    Infrared 2	  848x480	@ 60Hz	Y8
    Infrared 1	  848x480	@ 60Hz	Y8
    Infrared 2	  848x480	@ 30Hz	Y8
    Infrared 1	  848x480	@ 30Hz	Y8
    Infrared 2	  848x480	@ 15Hz	Y8
    Infrared 1	  848x480	@ 15Hz	Y8
    Infrared 2	  848x480	@ 6Hz	Y8
    Infrared 1	  848x480	@ 6Hz	Y8
    Infrared 2	  640x480	@ 90Hz	Y8
    Infrared 1	  640x480	@ 90Hz	Y8
    Infrared 2	  640x480	@ 60Hz	Y8
    Infrared 1	  640x480	@ 60Hz	Y8
    Infrared 1	  640x480	@ 30Hz	Y8
    Infrared 2	  640x480	@ 30Hz	Y8
    Infrared 1	  640x480	@ 15Hz	Y8
    Infrared 2	  640x480	@ 15Hz	Y8
    Infrared 2	  640x480	@ 6Hz	Y8
    Infrared 1	  640x480	@ 6Hz	Y8
    Infrared 1	  640x400	@ 25Hz	Y16
    Infrared 2	  640x400	@ 25Hz	Y16
    Infrared 1	  640x400	@ 15Hz	Y16
    Infrared 2	  640x400	@ 15Hz	Y16
    Infrared 2	  640x360	@ 90Hz	Y8
    Infrared 1	  640x360	@ 90Hz	Y8
    Infrared 1	  640x360	@ 60Hz	Y8
    Infrared 2	  640x360	@ 60Hz	Y8
    Infrared 1	  640x360	@ 30Hz	Y8
    Infrared 2	  640x360	@ 30Hz	Y8
    Infrared 2	  640x360	@ 15Hz	Y8
    Infrared 1	  640x360	@ 15Hz	Y8
    Infrared 1	  640x360	@ 6Hz	Y8
    Infrared 2	  640x360	@ 6Hz	Y8
    Infrared 2	  480x270	@ 90Hz	Y8
    Infrared 1	  480x270	@ 90Hz	Y8
    Infrared 1	  480x270	@ 60Hz	Y8
    Infrared 2	  480x270	@ 60Hz	Y8
    Infrared 1	  480x270	@ 30Hz	Y8
    Infrared 2	  480x270	@ 30Hz	Y8
    Infrared 2	  480x270	@ 15Hz	Y8
    Infrared 1	  480x270	@ 15Hz	Y8
[... 211 lines cut ...]
    Color	  320x240	@ 30Hz	YUYV
    Color	  320x240	@ 6Hz	RGB8
    Color	  320x240	@ 6Hz	Y16
    Color	  320x240	@ 6Hz	BGRA8
    Color	  320x240	@ 6Hz	RGBA8
    Color	  320x240	@ 6Hz	BGR8
    Color	  320x240	@ 6Hz	YUYV
    Color	  320x180	@ 60Hz	RGB8
    Color	  320x180	@ 60Hz	Y16
    Color	  320x180	@ 60Hz	BGRA8
    Color	  320x180	@ 60Hz	RGBA8
    Color	  320x180	@ 60Hz	BGR8
    Color	  320x180	@ 60Hz	YUYV
    Color	  320x180	@ 30Hz	RGB8
    Color	  320x180	@ 30Hz	Y16
    Color	  320x180	@ 30Hz	BGRA8
    Color	  320x180	@ 30Hz	RGBA8
    Color	  320x180	@ 30Hz	BGR8
    Color	  320x180	@ 30Hz	YUYV
    Color	  320x180	@ 6Hz	RGB8
    Color	  320x180	@ 6Hz	Y16
    Color	  320x180	@ 6Hz	BGRA8
    Color	  320x180	@ 6Hz	RGBA8
    Color	  320x180	@ 6Hz	BGR8
    Color	  320x180	@ 6Hz	YUYV

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

@JRauer  From the result of rs-enumerate, 640x480 is supported for your setup. Did you see the issue when both depth and color streaming enabled with 640x480@30fps or you also got the issue when enable only depth or color?

JRauer, 2019-03-23:
@RealSenseCustomerSupport Color streaming with these parameters is no problem. Only the depth camera does not work with 640x480@30fps independent of color stream being activated or not, when I try to launch it via the python or ros api as well as via realsense-viewer.

RealSenseCustomerSupport, 2019-03-25:
----------------------------------------------

@JRauer  Could you please follow https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q&A#q-how-do-i-enable-librealsense-logs to enable and collect the logs for 640x480@30fps depth streaming issue? And any error messages pop up in the GUI? If yes, please also provide one snapshot. Thanks!

JRauer, 2019-03-25:
@RealSenseCustomerSupport I made further test and can now say that the issue occurs independent of the frame rate. I enabled Log-Level "Debug" and copied the messages here. After launching the camera with 640x480 I have to replug it to make it work with another resolution - if that helps determining the problem.

When launching **realsense-viewer** I get the following:
 25/03 14:03:24,746 WARNING [139694398494464] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
 25/03 14:03:35,129 ERROR [139694381709056] (types.h:214) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 25/03 14:03:35,175 ERROR [139694381709056] (error-handling.cpp:66) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 25/03 14:03:40,252 WARNING [139694756518528] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
 25/03 14:05:43,362 WARNING [139694281062144] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds

When launching **Ros-node** I get the following:
started roslaunch server http://localhost:35875/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/enable_t265: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
[... 464 lines cut ...]
[DEBUG] [1553518774.418967540]: Frameset contain (Color, 0, RGB8) frame. frame_number: 0 ; frame_TS: 12520.758000 ; ros_TS(NSec): 1553518774418296483
[DEBUG] [1553518774.419014645]: Not points
[DEBUG] [1553518774.419073464]: publishFrame(...)
 25/03 13:59:34,419 DEBUG [139806802913024] (archive.cpp:288) CallbackFinished,Color,0,DispatchedAt,1553518774419.200195
 25/03 13:59:34,419 DEBUG [139806802913024] (archive.cpp:288) CallbackFinished,Color,0,DispatchedAt,1553518774419.264160
 25/03 13:59:34,448 DEBUG [139806802913024] (sensor.cpp:453) FrameAccepted,Color,Counter,1,Index,0,BackEndTS,1553518774417.587891,SystemTime,1553518774448.863770 ,diff_ts[Sys-BE],31.275879,TS,12554.118000,TS_Domain,Hardware Clock,last_frame_number,0,last_timestamp,12520.758000
 25/03 13:59:34,449 DEBUG [139806802913024] (archive.cpp:502) CallbackStarted,Color,1,DispatchedAt,1553518774449.760254
 25/03 13:59:34,449 DEBUG [139806802913024] (sync.cpp:143) DISPATCH (TS: )--> Color 1 12554.118000 

 25/03 13:59:34,449 DEBUG [139806802913024] (sync.cpp:143) DISPATCH (TS: (FN: I Depth I Infrared I Infrared ) I Color )--> Color 1 12554.118000 

[... cut ...]
 25/03 13:59:34,450 DEBUG [139806802913024] (archive.cpp:288) CallbackFinished,Color,0,DispatchedAt,1553518774450.122314
 25/03 13:59:34,450 DEBUG [139806802913024] (archive.cpp:295) Frame Callback [Color#0] overdue. (Duration: 1553518774450.122314ms, FPS: 30, Max Duration: 32ms)
 25/03 13:59:34,450 DEBUG [139806802913024] (syncer-processing-block.cpp:28) SYNCED: Color 1, 12554.118000 
 25/03 13:59:34,450 DEBUG [139806802913024] (archive.cpp:502) CallbackStarted,Color,1,DispatchedAt,1553518774450.223633

RealSenseCustomerSupport, 2019-03-27:
----------------------------------------------

@JRauer  What's the kernel version and any kernel patch did you apply? Can the camera work properly with Linux RealSenseViewer for 640x480@30fps depth only?

JRauer, 2019-03-29:
@RealSenseCustomerSupport No, the camera does also not work with RealSenseViewer with these parameters giving "REC Error". The Kernel version is Linux 4.10.0.28-generic

RealSenseCustomerSupport, 2019-04-01:
----------------------------------------------

@JRauer That's really weird! Did you apply the kernel patch as the script "./scripts/patch-realsense-ubuntu-lts.sh"?

RealSenseCustomerSupport, 2019-04-04:
----------------------------------------------

@JRauer Any update for the issue? Looking forward to your update. Thanks!

JRauer, 2019-04-04:
@RealSenseCustomerSupport I'll keep you informed, but I am not in the office until middle of next week. I have not tried installing the patch. Since I don't necessarily need 640x480 and higher resolutions are working fine: is it possible to harm the working resolutions by installing the patch? If so, I am not sure if i want to risk getting new issues by trying to get something to work, I don't really need.

RealSenseCustomerSupport, 2019-04-04:
----------------------------------------------

@JRauer I don't think the patch will harm the working resolutions. And if you're ok with the higher resolution, then you can keep the current configuration.

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

@JRauer Any other questions for this ticket? If no, will close it accordingly.

JRauer, 2019-04-08:
@RealSenseCustomerSupport no, thank you very much for your support!

closed 2019-04-08 by JRauer

rupak-d, 2021-03-16:
Had a similar issue where 480x270 and 848x480 resolutions worked, but 640x360 did not. Worked after doing an on chip calibration.

rupak-d, 2021-03-17:
Hi @MartyG-RealSense, do we know why this happens? And why does the on-chip calibration fix this?

MartyG-RealSense, 2021-03-17:
There was a similar case in October 2020 where all depth resolutions except 848x480 were available.  In that case, the RealSense SDK's other calibration tool, the **Dynamic Calibrator**, corrected the problem.

Given that a recalibration with either calibration tool (On-Chip or Dynamic) made the missing resolutions available again, writing a new calibration to the camera hardware's internal storage may have corrected a glitch in the previously stored calibration data.

rupak-d, 2021-03-18:
Is there a way to know that there are some resolutions that are unavailable or there's a glitch in the calibration data?

MartyG-RealSense, 2021-03-18:
@rupak-d   If you wish to check whether a specific stream profile is currently available (supported) then you could check the requested stream configuration with the **can_resolve()** instruction to see whether the request can be fulfilled.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1config.html#a4ff9ae18bf1b3aaaaaf9bba524b04d2c

You can also retrieve all of the currently available stream profiles with the instruction **get_stream_profiles()**
and list their details:

https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/api_how_to.h#L349

I am not aware of a function that can specifically detect a calibration table corruption and inform of it.  An automated workaround may be to test for the availability of a particular stream profile and activate code to write a new calibration table to the camera if the requested profile cannot be provided.

rupak-d, 2021-03-24:
The stream profiles show up even when the cameras do not run at that resolution.
Is there a way to recreate this consistently?

MartyG-RealSense, 2021-03-24:
Due to the high rarity of the problem of certain resolutions becoming unavailable, I do not have details of how to test for it.

In regard to testing for calibration data corruption, a RealSense user described in detail their calibration testing process in the link below.

https://github.com/IntelRealSense/librealsense/issues/5606