realsenseai/librealsense #3377: Windows 10 doesn't detect realsense

freaad, 2019-03-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {T265 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10 } | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  {  2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
My windows desktop and laptop doesn't recognize the my new T265 realsense. Those are worked well with 415 but it is not working with T265. When I connect T265 it doesn't give correct sound and it just give " we do not know what is this machine" sound. Do you have any idea?

dorodnic, 2019-03-05:
Hi @freaad 
Did you try with the latest [RealSense Viewer](https://github.com/IntelRealSense/librealsense/releases/download/v2.18.1/Intel.RealSense.Viewer.exe)?
Prior to opening the Viewer the T265 should appear in Device Manager under Universal Serial Bus Devices as Movidious MA2X5X, and once you open the Viewer (or any other SDK app) it should become Intel RealSense Tracking Camera T265

freaad, 2019-03-06:
It is now working. Thank you.

closed 2019-03-06 by freaad

dorodnic, 2019-03-06:
Everything is under [releases](https://github.com/IntelRealSense/librealsense/releases). We published 2.19 with more features yesterday.
Depth quality tool will not do anything useful with the T265, as the name implies, but you should be able to work with the device in python via pip.