realsenseai/librealsense #3379: T265 fisheye stream intrinsics

AmadeusSVX, 2019-03-05:
Required Info |  
-- | --
Camera Model | T265
Operating System & Version | Windows 10
Platform | PC x64
SDK Version | 2.18.1 
Language | C#
Segment | PC

I'm trying to use fisheye streams of T265.
Now they are available as streams. But I cannot get intrinsics of them from their profiles as depth/infrared streams of D4xx series. 

I'm trying with below code. There is no error returned, but l_intrinsic and r_intrinsic values are all 0.
Do I need to do with something different manner or simply not implemented to expose their values yet?

	rs2::pipeline_profile selection = pipe.start(cfg);

	auto left_fisheye_stream = selection.get_stream(RS2_STREAM_FISHEYE, 1).as<rs2::video_stream_profile>();
	auto right_fisheye_stream = selection.get_stream(RS2_STREAM_FISHEYE, 2).as<rs2::video_stream_profile>();
	auto l_intrinsic = left_fisheye_stream.get_intrinsics();
	auto r_intrinsic = right_fisheye_stream.get_intrinsics();

dorodnic, 2019-03-05:
Hi @AmadeusSVX 
No, this should work, we need to check why it doesn't.

honpong, 2019-03-11:
@dorodnic  I
If I get_instrinsics(), I got all 0 instrinsics and after that the wait_for_frames() will throw exception. 
If I don't query instrincis, I have no problem wait_for_frames().
I am using Windows.

RealSenseCustomerSupport, 2019-03-12:
----------------------------------------------

I can observe this issue based on the modified version of the rs-enumerate-devices tool. This issue needs to be addressed internally.

seiya0412, 2019-03-18:
Just FYI (and for making this thread active).
This issue can be reproduced on Ubuntu 16.04 with SDK 2.19.0.

bfulkers-i, 2019-03-20:
This is being worked on in https://github.com/IntelRealSense/librealsense/pull/3541 and once that PR is merged the T265 intrinsics should be exposed. See also #3544 #3497

bfulkers-i, 2019-04-10:
intrinsics and extrinsics are now exposed in librealsense `2.20.0`, you can see an example using both to render a virtual object in the fisheye frame here:

https://github.com/IntelRealSense/librealsense/tree/master/examples/ar-basic

Can I close this issue?

closed 2019-05-08 by dorodnic