realsenseai/librealsense #3380: rs2::process_interface can not be found

jinfagang, 2019-03-05:
There are not such interface in realsense SDK:

[image]

Build error:

```
error: ‘rs2::process_interface’ has not been declared
     filter_options(const std::string name, rs2::process_interface &filter);
                                                 ^~~~~~~~~~~~~~~~~

```

Did I miss something to include? (I have include almost everything)

jinfagang, 2019-03-05:
@BarIntel @ofirintel @kfirosIntel This is very emergency now. As it is shown above, the interface can not found and I can not apply built in filter algorithm in side sdk on depth image.

Does there any advices about this? Or what should I do to deal with post-processing make depth much more stable and not wavely (reduce noise so on..)?

dorodnic, 2019-03-05:
This is captured in API changes between [2.16 to 2.17](https://github.com/IntelRealSense/librealsense/wiki/API-Changes#renamed).
You can also review [examples/post-processing](https://github.com/IntelRealSense/librealsense/tree/master/examples/post-processing) for a working sample with the new API.

jinfagang, 2019-03-05:
@dorian3d Thank u for you reply. I think the example code does not update yet. I am just also think it might  changed into porcess_block, but that does not has a method called `process`, a `start` method only which does not return anything

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

@jinfagang Could you please double check https://github.com/IntelRealSense/librealsense/tree/master/examples/post-processing again?

RealSenseCustomerSupport, 2019-03-26:
----------------------------------------------

@jinfagang Any other questions about this ticket? Please update. Thanks!

RealSenseCustomerSupport, 2019-04-01:
----------------------------------------------

@jinfagang If no other questions, will close the ticket accordingly. Thanks!

closed 2019-04-08 by RealSenseCustomerSupport