realsenseai/librealsense #3381: RealSense Viewer not working when camera connected

anembac, 2019-03-05:
| | |
|---|---|
|**librealsense**|2.18.1 RELEASE|
|**OS**|Windows 10|

RealSense Viewer only launches when my D435 isn't connected. When connected it opens the GUI but is completely unresponsive. (can't click any buttons etc.)

Windows does find the device when it is connected and confirms it's connected through a USB 3.0 port.

MartyG-RealSense, 2019-03-05:
There was a similar case to this in January 2019.  It ended up being closed without a clear solution, but I will quote it here so the Intel guys on this forum can refer back to it.

https://github.com/IntelRealSense/librealsense/issues/3148

Firifire, 2019-03-05:
Well the 2.18.1 & 0 Viewer doesn't not launch at all for me, regardless the camera is connected or not.
build 2.17.1 works fine for me

CaleMcCollough, 2019-03-05:
I'm using 2.18.1 and I'm able to view the depth sensor feed but once I tried to view the camera feed the window goes unresponsive white and crashes.

dorodnic, 2019-03-05:
These are somewhat unrelated issues,
The Access Denied problem on Windows is related to Windows rolling updates, and seem to be a bug on Microsoft side. We have encountered it (but not enough for a proper root cause) and so far it seem to be ultimately solved by a sequence of driver and windows updates.
The issue of 2.18.1 not launching is most likely related to graphics (GPU) drivers, can you please elaborate about the setup?
The 3rd issue is something new. I can suggest trying the [official drivers for Windows](https://downloadcenter.intel.com/download/27799/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10)

Firifire, 2019-03-05:
My setup is
GTX 850M (419.35)
i7-4710 HQ
Intel HD Graphics 4600 (15.40.41.5058)

anembac, 2019-03-05:
Interestingly I am also using a GTX 850M on 419.35.

GTX 850M (419.35)
i5-4200H
Intel HD Graphics 4600 (20.19.15.4549)

Firifire, 2019-03-06:
I just updated to Windows Build 1809. And I no longer have issues launching these application, including 2.19.0

1546926687, 2019-03-13:
I have just encountered the same problem. I tried to download different versions (2.18,2.19) but it didn't work.

closed 2019-03-16 by RealSenseCustomerSupport

tobecker, 2019-05-08:
I have the same problem using version 2.20.0 and 2.21.0 with Windows 10 1809 with Intel i7-6900K / Nvidia GTX960 (Driver 430.39).

With a different PC with AMD Ryzen 2400G (also Windows 10 1809) and internal GPU, version 2.20.0 and 2.21.0 runs perfect.

SarathVakacharla, 2019-06-19:
> changing data privacy options
Where to change the data privacy options?

MartyG-RealSense, 2019-06-19:
@SarathVakacharla  If you type 'privacy' into the text box at the bottom of the Windows screen (next to the button where the PC is shut down), you will find the 'Privacy Settings' option.  Select the "Camera" category from the left-side options panel, and ensure that 'Allow access to the camera on this device' and 'Allow apps to access your camera' are both set to "On".

Additionally, if you scroll further down, you can set which specific programs are allowed to access the camera.

[image]

SarathVakacharla, 2019-06-19:
I had this on, but I was still getting this error on Windows 10 1809 build.
Upgrading librealsense from 2.19.x to 2.21.0 and system restart helped.

MartyG-RealSense, 2019-06-19:
Glad you found a solution!