realsenseai/librealsense #3383: Does the align function have depth_optical_frame choice?

QiuYilin, 2019-03-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I found that the frame of pointcloud is camera_depth_optical_frame. But I do the calibration with librealsense api and there is no depth_optical_frame to choose. I have to unify the frames.

RealSenseCustomerSupport, 2019-03-12:
----------------------------------------------

Hi @QiuYilin,

Can you elaborate more detail?
1. which librealsense and ros wrapper versions do you use?
2. You have your own intrinsic/extrinsic custom calibration using librealsense API?
3. which SR300 firmware version?
4. did you try to run "demo_pointcloud.launch"?

RealSenseCustomerSupport, 2019-04-02:
----------------------------------------------

Hi @QiuYilin,

Still need any support for this topic. If not, will close it.

closed 2019-05-09 by RealSenseCustomerSupport