realsenseai/librealsense #3384: T265 camera can not be opened in realsense-viewer app.

chunyang-zhang, 2019-03-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Realsense T265                     | 
| Firmware Version                   | Unknown version                   | 
| Operating System & Version  |   Ubuntu 16.04                       | 
| Kernel Version (Linux Only)   |  4.4.0-142                              | 
| Platform                                 | x64                                         |
| SDK Version                          |  librealsense 2.18.1                | 
| Language                              |  C++                                       | 
| Segment			        |  Robot/others                         | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I have installed RealSense SDK (version 2.18.1) in my OS successfully. I can use the installed tools(realsense-viewer) to open the cameras successfully such as D435 , D435i without any errors, but T265 failed. So I tried to reinstalled the SDK by apt tool according to "distribution_linux.md" , but the problem  remained. How do I open the camera T265 correctly in my machine?

I use the terminal command to detect the device ：
$ rs-enumerate-devices 
 05/03 20:31:34,326 ERROR [140514671720256] (backend-v4l2.cpp:518) Cannot access /sys/class/video4linux
No device detected. Is it plugged in?

but camera's data line is actually plugged in.

the terminal command:
$ modinfo uvcvideo | grep "version:"
version:        1.1.2.realsense-1.3.4
srcversion:     D08235669143A1AC9F127F5

the terminal command:
$ lsusb 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 001 Device 009: ID 03e7:2150  
Bus 001 Device 003: ID 1a2c:0044 China Resource Semico Co., Ltd 
Bus 001 Device 002: ID 1c4f:0002 SiGma Micro Keyboard TRACER Gamma Ivory
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

the line "Bus 001 Device 009: ID 03e7:2150" is found when I connect the camera to my computer.

So, How can I use the T265 correctly? thank you !

ev-mp, 2019-03-05:
@chunyang-zhang ,
When compiling SDK from sources for T265 you **must** set CMake's BUILD_WITH_TM2=ON. 
You also **must** have an un-obscured internet access (mitigate/verify proxy) for CMake to download the required TM2 firmware blocks.
if those conditions are not met the compiled module will not support T265. You'll also get a warning posted by Cmake

Note - all the official SDK distributions for Librealsense (Debian/Win/Python) have the T265 support built-in.

Also check a similar report in #3361

chunyang-zhang, 2019-03-06:
@ev-mp, Thank you a lot, it works!

closed 2019-03-16 by RealSenseCustomerSupport

zssjh, 2019-06-18:
I met the same problem and test output with @chunyang-zhang ,  I set  CMake's BUILD_WITH_TM2=ON and download the required TM2 firmware blocks successfully. There still nothing output on realsense-viewer?   @ev-mp @RealSenseCustomerSupport Many thanks!