realsenseai/librealsense #3385: Unrooted Android: How to build librealsense.so and how to interface

xtrawurst, 2019-03-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version |  Android | 
| Language                            |  C++ | 

### Issue Description
With PR #3083 there is support for unrooted Android on the development branch. However the only 
[how-tos](https://github.com/matkatz/librealsense/tree/unrooted_android_device_support/wrappers/android) [out there](https://github.com/matkatz/librealsense/blob/unrooted_android_device_support/doc/android.md) are for Android Studio apps.

We are using Qt5, not Android Studio for development. Therefore we need to get the raw librealsense.so / librealsense.a file that we can integrate into our project.

In the ["old" Android docs](https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidNativeSamples.md) the build was through CMake and indeed produced a librealsense.so / librealsense.a file in the end. Is this still supported in the unrooted model (What would be the CMake variables to set? What are the necessary source files to build librealsense.so for unrooted Android?)?

If yes, I assume there are some necessary steps to integrate with the Java-based USB management. What are the interface points there? Is there a minimal example that shows how to get something going on Android, just using librealsense.so?

dorodnic, 2019-03-05:
Hi @xtrawurst 
I'm not entirely familiar with Qt for Android. First things I wanted to note is that we are not removing the "old" method of working on Android, it is still being covered by CI and should continue to work.
The new approach relies on requesting proper permissions in Java for the activity and passing that information to C++. You can build librealsense2.so via CMake with the new backend (by setting a dedicated CMake flag) but you'd need a way to handle permissions in order for it to work on an unrooted phone. 
@matkatz might be able to offer more useful answer

xtrawurst, 2019-03-05:
@dorodnic Thanks!

@matkatz No specific Qt instructions are necessary. Just some instructions on how to best build librealsense2.so for Android and how the Java-based permissions connect with the native stuff. Also, I know that in the past libusb was used (for Android, too). IIRC all the USB communication is now through direct ioctl() rather than libUSB. But is libUVC still used on Android? I saw some files in the PR that carried UVC in the name so I was wondering. It would be good to have a high-level overview of how things work now on Android.

indigohedgehog, 2019-03-29:
@xtrawurst It seems that librealsense is designed to be linux-compatible, thus using low-level libraries as libsusb, which is also supported on Android, but with a permission drawback. A better approach would be to have the devices to provide support for Android Open Accessory, so we could instead use USBManager. 
For instance other imaging devices are currently following that approach, like Flir One.

daniel-piro, 2019-03-30:
@indigohedgehog Android native support has been added to librealsense (since V2.19.0). It is built on top of Android's UsbHost API and works on unrooted devices. 

A sample Java project with  can be found here with instructions (readme.md):
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/java_example

zhouwg, 2019-04-02:
@xtrawurst, there is an Android wrapper could be found in Github:https://github.com/zhouwg/irsa.wish it's useful for your purpose.

matkatz, 2019-04-10:
Hi @xtrawurst 
Sorry that it took so long, as @dorodnic said, due to Android permissions limitations, before you can start work with either Java or Native code, you must request permissions for the USB device.
The java wrapper contains 3 classes (Enumerator, DeviceWatcher, RsContext) which together detect USB attach events, request for user permissions and create a UVC device on the native said via JNI calls.
Once the UVC device is created you can either use the c++ API or the Java Api.
Building the so without the AAR means that you will need to handle the USB permissions and create the UVC devices on the native said.

RealSenseCustomerSupport, 2019-05-03:
----------------------------------------------

Closing this ticket, please feel free to reopen if problem persists. Thank you.

closed 2019-05-10 by RealSenseCustomerSupport