realsenseai/librealsense #3386: T265 not detected on macOS with realsense-viewer

dataplayer12, 2019-03-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  T265 | 
| Firmware Version                   | #None (see below) | 
| Operating System & Version | MacOS  | 
| Platform                                 | Mac |
| SDK Version                            |  2.17.0 }                         | 
| Language                            |  C++/python                          | 
| Segment			|  others                          | 

The T265 camera can not be detected on MacOS with realsense-viewer. I have tried all the other examples which compile with librealsense. I have also tried accessing the camera via C++ and Python scripts, but to no avail.

Running the python example for T265 from [here](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_example.py) gives this traceback:

```python

>>>python2 t265_example.py 
t265_example.py:12: FutureWarning: pybind11-bound class 'pyrealsense2.processing_block' is using an old-style placement-new '__init__' which has been deprecated. See the upgrade guide in pybind11's docs. This message is only visible when compiled in debug mode.
  import pyrealsense2 as rs
t265_example.py:12: FutureWarning: pybind11-bound class 'pyrealsense2.filter' is using an old-style placement-new '__init__' which has been deprecated. See the upgrade guide in pybind11's docs. This message is only visible when compiled in debug mode.
  import pyrealsense2 as rs
Traceback (most recent call last):
  File "t265_example.py", line 22, in <module>
    pipe.start(cfg)
RuntimeError: No device connected
```

I can't know the firmware of the camera since I can't connect to it in any way, but I imagine there is only one firmware version in production at the moment as T265 has just been launched. What am I doing wrong and how to correct it?

dorodnic, 2019-03-05:
T265 is not currently supported on Mac-OS, as documented on the site and mentioned on GitHub. 
We are working on adding this functionality as soon as possible, but for now the T265 can work on Mac using Parallels Ubuntu image.

dataplayer12, 2019-03-05:
Thank you for the clarification. I am closing the issue then. I hope Mac support will be added soon

closed 2019-03-05 by dataplayer12

dorodnic, 2019-03-06:
@dataplayer12 - I created a pull-request with the work-in-progress we have towards Mac support, hope you might find it useful

dataplayer12, 2019-03-06:
Thank you for the PR. I pulled the PR and compiled it with `-DBUILD_EXAMPLES=true`, which made the apps. I ran `rs-enumerate-devices` and it could not detect the T265 camera.

```Shell
No device detected. Is it plugged in?
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
```

Running the python script as before, now shows a different error.
```python
>>>python2 t265_example.py 
t265_example.py:12: FutureWarning: pybind11-bound class 'pyrealsense2.processing_block' is using an old-style placement-new '__init__' which has been deprecated. See the upgrade guide in pybind11's docs. This message is only visible when compiled in debug mode.
  import pyrealsense2 as rs
t265_example.py:12: FutureWarning: pybind11-bound class 'pyrealsense2.filter' is using an old-style placement-new '__init__' which has been deprecated. See the upgrade guide in pybind11's docs. This message is only visible when compiled in debug mode.
  import pyrealsense2 as rs
Traceback (most recent call last):
  File "t265_example.py", line 27, in <module>
    frames = pipe.wait_for_frames()
RuntimeError: Frame didn't arrived within 5000
```
Does this mean that the camera is detected by `pyrealsense`? Am I doing something wrong? Have you verified that this patch works on mac? Thanks a lot for your efforts!!

dorodnic, 2019-03-06:
Did you try the Viewer?
Yes, I tested using the Viewer, `rs-capture` and `rs-pose` on an 2017 MacBook Pro. 
One caveat of the build process is that it requires internet connection. Look for:
```
-- Checking internet connection...
-- Internet connection identified, enabling BUILD_WITH_TM2
```
In Cmake output. If it fails, it will disable T265 support. However, if all is well, you should see:
```
-- T265 Product versions:
-- - HOST 0.19.3.1711 (Default from versions.cmake)
-- - Remote FW 0.0.18.5129 (Default from versions.cmake)
-- - Remote CENTRAL APP 2.0.19.271 (Default from versions.cmake)
-- - Remote CENTRAL BL 1.0.1.112 (Default from versions.cmake)
```
(The reason this is needed is because T265 firmware is not stored on the device and cannot be managed inside the git repository, so we download it during CMake)
Once make is done, it is always a good idea to check that the realsense library being used is actually the one that was built and not some previous version.

dorodnic, 2019-03-06:
Feel free to reopen the ticket.
We are trying to support so many product, platform and language combinations that it necessitates prioritizing the more common ones first, but you should definitely have the option to develop on a platform that is convenient for you.

radfordi, 2019-11-08:
If you've had problems with T265 on a Mac previously, then #5213 will likely address them.

neilyoung, 2019-11-08:
If somebody, who has made a T265 run together with a Mac would share his build procedure, that would be awesome. I wasn't able to produce a makefile on Catalina yesterday... It's not in my focus, but I like my Mac more than my Ubuntu laptop...