realsenseai/librealsense #3387: T265 Drift

JBBee, 2019-03-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.4577 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                               | 
| Platform                                 |  |
| SDK Version                            | 2.18.1                         | 
| Language                            |  C++                        | 
| Segment			|  AR                         | 

### Issue Description
I just received the T265.  I plugged it in and opened the Viewer, turned on all streams and walked around the room in various patterns and wiggled the camera a bit.  The pose stream showed a large amount of drift.  After walking around the room and returning to the place that I started the position of the pose stream was no where near 0.  It was over 1 meter away.  With this amount of drift it would be unsuitable for our AR application.  

Is there any type of configuration that needs to be done before using the camera?

Thanks.

dorodnic, 2019-03-05:
Hi @JBBee 
Sorry to hear. 
There is not much configurability in the built-in SLAM as far as I know, but you could try the version on the development branch, it includes newer firmware version (the pre-built artifacts should be published soon as well) that might perform a bit better.

JBBee, 2019-03-05:
Hi @dorodnic 
The drift looks reasonable now on the development branch.  And I'm sure as you're developing it will continue to improve..  Thank you for this great product.

dorodnic, 2019-03-05:
Glad to hear!
We also had an internal discussion and I wanted to bring [static node APIs](https://github.com/IntelRealSense/librealsense/pull/3324) to your attention. This is very new and not sufficiently explained, but it can be a very useful capability for the AR use-case.
Basically it should let you create local coordinate systems to reduce error around specific points of interests.

jattiogbe, 2019-03-15:
Hi @JBBee 
I am looking for information to assess the utility of T265 in my positioning project. I'm not sure of how to exploit the data from the device, especially the trajectory.
Would you please export a trajectory recording and upload it so that I can check it out ?

I've already got a rosbag example that I use with RealSense Viewer, but I can't do anything but display the camera vision and 3D trajectory. I can't export any data out of it..

closed 2019-03-16 by RealSenseCustomerSupport