realsenseai/librealsense #3393: Pointcloud is very bad and wavely even using filters

jinfagang, 2019-03-06:
I am using some filters built in, but result are really wavely:

[image]

With the quality of point cloud, I don't think it can be used to do any applications. model is D435

dorodnic, 2019-03-06:
This type of errors can be usually significantly reduced by recalibrating the camera using [provided tools](https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html)

jinfagang, 2019-03-06:
@dorian3d Thanks for you reply. I am now using the right calibration now, it seems better:

[image]

As you can see, the plane can be flat now (though it is very close, but I am good with it)

**another serious thing is**: how to remove the points in the middle air?

you can tell it is full with points that doesn't  has any objects. How to remove those middle air points out?

[image]

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Hi @jinfagang,

Please refer to the following white-paper for tuning the depth performance. Choosing a higher accuracy preset and/or applying some filters would be something to quickly try and see the outcome using the RealSense Viewer app.
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

closed 2019-04-09 by RealSenseCustomerSupport