realsenseai/librealsense #3394: Build errors on Beaglebone Black

jdwolk, 2019-03-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | None | 
| Operating System & Version |   Ubuntu 16.04   | 
| Kernel Version (Linux Only)    |  4.4.155-ti-r151  | 
| Platform                                 | Beaglebone Black |
| SDK Version                            |  None  | 
| Language                            |  None   | 
| Segment			|  Robot   | 

### Issue Description

Hi,

I'm trying to build on a Beaglebone Black. I'm following the instructions [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md) and [here](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md##prerequisites). When I run `make` it fails (in this case I'm running with ``VERBOSE=1` flag):

```
[ 41%] Building CXX object CMakeFiles/realsense2.dir/src/media/playback/playback_sensor.cpp.o
/usr/bin/c++   -DBUILD_EASYLOGGINGPP -DBUILD_STATIC=1 -DELPP_NO_DEFAULT_LOG_FILE -DELPP_THREAD_SAFE -DHWM_OVER_XU -DRS2_USE_V4
L2_BACKEND -DUNICODE -DUSE_SYSTEM_LIBUSB -DWITH_TRACKING=1 -Drealsense2_EXPORTS -isystem /usr/include/libusb-1.0 -I/home/ubunt
u/Downloads/librealsense/realsense2 -I/home/ubuntu/Downloads/librealsense/src -I/home/ubuntu/Downloads/librealsense/third-part
y/realsense-file/rosbag/console_bridge/include -I/home/ubuntu/Downloads/librealsense/third-party/realsense-file/rosbag/cpp_com
mon/include -I/home/ubuntu/Downloads/librealsense/third-party/realsense-file/rosbag/rosbag_storage/include -I/home/ubuntu/Down
loads/librealsense/third-party/realsense-file/rosbag/roscpp_serialization/include -I/home/ubuntu/Downloads/librealsense/third-
party/realsense-file/rosbag/rostime/include -I/home/ubuntu/Downloads/librealsense/third-party/realsense-file/rosbag/roscpp_tra
its/include -I/home/ubuntu/Downloads/librealsense/third-party/realsense-file/rosbag/roslz4/include -I/home/ubuntu/Downloads/li
brealsense/third-party/realsense-file/rosbag/msgs -I/home/ubuntu/Downloads/librealsense/third-party/realsense-file/boost -I/ho
me/ubuntu/Downloads/librealsense/third-party/realsense-file/lz4/lib -I/home/ubuntu/Downloads/librealsense/include -I/home/ubun
tu/Downloads/librealsense/third-party/libtm/libtm/include  -fPIC -pedantic -g -Wno-missing-field-initializers -Wno-switch -Wno
-multichar -Wsequence-point -Wformat-security -mfpu=neon -mfloat-abi=hard -ftree-vectorize -pthread -fPIC   -std=gnu++11 -o CM
akeFiles/realsense2.dir/src/media/playback/playback_sensor.cpp.o -c /home/ubuntu/Downloads/librealsense/src/media/playback/pla
yback_sensor.cpp
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
CMakeFiles/realsense2.dir/build.make:566: recipe for target 'CMakeFiles/realsense2.dir/src/media/playback/playback_sensor.cpp.
o' failed
make[2]: *** [CMakeFiles/realsense2.dir/src/media/playback/playback_sensor.cpp.o] Error 4
make[2]: Leaving directory '/home/ubuntu/Downloads/librealsense/build'
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/realsense2.dir/all' failed
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
make[1]: Leaving directory '/home/ubuntu/Downloads/librealsense/build'
Makefile:127: recipe for target 'all' failed

```

Thanks in advance for any help!

dorodnic, 2019-03-06:
Hi @jdwolk 
We actually have a note regarding this error in [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md).
[Adding a swap file](https://linuxize.com/post/how-to-add-swap-space-on-ubuntu-18-04/) will solve the problem.

jdwolk, 2019-03-07:
@dorodnic got it, thanks for pointing that out. I was able to build by creating a swapfile using this:

```
cd /; sudo fallocate -l 1G /swapfile;
sudo chmod 600 /swapfile;
sudo mkswap /swapfile;
sudo swapon /swapfile;
sudo swapon -s;

Edit /etc/fstab, add:
/swapfile   none    swap    sw    0   0
```

I'm now having problems with `realsense-viewer` segfaulting + most of the `rs-` example commands are returning "Frame didn't arrived within 5000". I'll close this issue since I was able to get the build working though.

closed 2019-03-07 by jdwolk