realsenseai/librealsense #3395: load *.rssdk files with python

zhouyongxiu, 2019-03-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Operating System & Version |    MacOS  | 
| Platform                                 | PC |
| Language                            |  python                           | 
| Segment			|  others                           | 

### Issue Description
I have lots of files save as *.rssdk from SR300, how can I load those files and convert the stream to numpy with opencv? I known it is good suggestion to save file as *.bag, but I only have *.rssdk files, or is there any method to convert *.rssdk to *.bag.

MartyG-RealSense, 2019-03-06:
You have already seen and commented on the previous discussion on this topic.

https://github.com/IntelRealSense/librealsense/issues/2660

There was a discussion amongst users in the RealSense community about converting RSSDK files in Librealsense 2.0 using a 'grabbing' program called **RealSense2Grabber**.  Community members such as @UnaNancyOwen were involved in this discussion.

https://github.com/PointCloudLibrary/pcl/pull/2214

https://github.com/PointCloudLibrary/pcl/pull/2214/files/33610a4b3a9864953200cc6eda6d15d379a87f5e

This effort to create a grabber for Librealsense2 was inspired by an earlier RealSenseGrabber in 2016 for Librealsense1 (Legacy Librealsense).

https://github.com/lebronzhang/pcl/pull/14

It is a long and involved discussion.  I believe that the end-result of it was the **realsense2_grabber.h** script in the repository linked to below.

https://rgit.acin.tuwien.ac.at/v4r/v4r/blob/ff3fbd6d2b298b83268ba4737868bab258262a40/modules/io/include/v4r/io/realsense2_grabber.h

I know it is not based on Python, but this may be the closest thing available to a ready-made solution.

RealSenseCustomerSupport, 2019-03-25:
----------------------------------------------

Hi zhouyongxiu,

Given what MartyG-RealSense has commented here, do you have any updates or do you still need help?

Thanks!

RealSenseCustomerSupport, 2019-04-09:
----------------------------------------------

Hi zhouyongxiu,

If there is nothing else needed, this one will be closed soon.

Thanks!

closed 2019-04-19 by RealSenseCustomerSupport