realsenseai/librealsense #3396: t265 pos data error on t265_example.py

chompanyi, 2019-03-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | t265| 
| Firmware Version                   | Real Sense Viewer 2.19.0 | 
| Operating System & Version |   Window 10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Laptop  |
| SDK Version                            | 2.0                        | 
| Language                            |  python                          | 
| Segment			| Robot                 | 

### Issue Description
running example on librealsense/wrappers/python/examples/t265_example.py

AttributeError: "'pyrealsense2.pyrealsense2.composite_frame' object has no attribute 'get_pose_frame'"

Where is API doc?

ev-mp, 2019-03-06:
Hello @chompanyi , this demo was updated in v2.19.0.
Please make sure that you're using the latest pyrealsense.

chompanyi, 2019-03-06:
I am using latest demo v2.19.0

#!/usr/bin/python
# -*- coding: utf-8 -*-
## License: Apache 2.0. See LICENSE file in root directory.
## Copyright(c) 2019 Intel Corporation. All Rights Reserved.

#####################################################
##           librealsense T265 example             ##
#####################################################

# First import the library
import pyrealsense2 as rs

# Declare RealSense pipeline, encapsulating the actual device and sensors
pipe = rs.pipeline()

# Build config object and request pose data
cfg = rs.config()
cfg.enable_stream(rs.stream.pose)

# Start streaming with requested config
pipe.start(cfg)

try:
    for _ in range(50):
        # Wait for the next set of frames from the camera
        frames = pipe.wait_for_frames()

        # Fetch pose frame
        pose = frames.get_pose_frame()<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
        if pose:
            # Print some of the pose data to the terminal
            data = pose.get_pose_data()
            print("Frame #{}".format(pose.frame_number))
            print("Position: {}".format(data.translation))
            print("Velocity: {}".format(data.velocity))
            print("Acceleration: {}\n".format(data.acceleration))

finally:
    pipe.stop()

chompanyi, 2019-03-06:
another error 
Traceback (most recent call last):
  File "C:/codebase/example1.py", line 32, in <module>
    data = pose.get_pose_data()
AttributeError: 'pyrealsense2.pyrealsense2.frame' object has no attribute 'get_pose_data'

P

chompanyi, 2019-03-06:
another error
Traceback (most recent call last):
File "xxxxxxxx/example1.py", line 32, in 
data = pose.get_pose_data()
AttributeError: 'pyrealsense2.pyrealsense2.frame' object has no attribute 'get_pose_data'

closed 2019-03-06 by chompanyi