realsenseai/librealsense #3397: How to manually reset 6DOF tracking of RealSense T265

UnaNancyOwen, 2019-03-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265                                                       | 
| SDK Version                            | 2.18.1                                                     | 
| Language                               | C++                                                        | 

### Issue Description
This is question.
How to manually reset 6DOF tracking of RealSense T265 on code?

```cpp
// Set Config
rs2::config config;
config.enable_stream( rs2_stream::RS2_STREAM_POSE, rs2_format::RS2_FORMAT_6DOF );

// Start Pipeline
rs2::pipeline pipeline;
rs2::pipeline_profile pipeline_profile = pipeline.start( config );

/* ... */

// Reset Pose
/* how to reset 6dof tracking? */
```

dorodnic, 2019-03-06:
Hi @UnaNancyOwen 
I think `pipeline.stop(); pipeline.start()` should do it

UnaNancyOwen, 2019-03-06:
@dorodnic Thanks, It works well.
```cpp
// Reset Pose
pipeline.stop();
std::this_thread::sleep_for( std::chrono::seconds( 1 ) ); // wait a little bit
pipeline.start();
```

closed 2019-03-06 by UnaNancyOwen

carlosefrias, 2020-03-04:
How to do this (Reset the pose)

`
// Reset Pose
pipeline.stop();
std::this_thread::sleep_for( std::chrono::seconds( 1 ) ); // wait a little bit
pipeline.start();
`
On the Unity wrapper?
Thanks in advance

tgaldi, 2020-07-20:
I use the following, where GetRSDevice() returns the active RSDevice component.

```
        private IEnumerator ResetTracker()
        {
            ApplicationControl.instance.GetRSDevice().enabled = false;

            yield return new WaitForSeconds( 1f );

            ApplicationControl.instance.GetRSDevice().enabled = true;

            ResetRoutine = null;
        }
```

Off topic but have you found that Unity either hangs or becomes unresponsive? This tends to happen when hitting stop. Setting a break point it seems related to the call to  m_pipeline.Dispose() in RsDevice's OnDisable(). I've tried both multithreaded and unity thread.

lms0, 2020-10-02:
> @dorodnic Thanks, It works well.
> 
> ```c++
> // Reset Pose
> pipeline.stop();
> std::this_thread::sleep_for( std::chrono::seconds( 1 ) ); // wait a little bit
> pipeline.start();
> ```

@UnaNancyOwen why do you add a delay of 1 second in between?