realsenseai/librealsense #3398: There's no VS2010-15.Integration.MD.props or VS2010-15.Integration.MT.props file

bingws, 2019-03-06:
I installed the SR300 SDK, and trying to set up a C++ develop environment, but there's no VS2010-15.Integration.MD.props or VS2010-15.Integration.MT.props file on my entire computer

MartyG-RealSense, 2019-03-06:
In my installation of the 2016 R2 SDK, the props files can be found at this SDK folder location:

C: > Program Files (x86) > Intel > RSSDK > props

[image]

bingws, 2019-03-06:
@MartyG-RealSense I found my RSSDK path: "C:\Windows\System32\config\systemprofile\AppData\Local\Intel\RSSDK", but nothing exists.
Which one did you install?I am installing "Intel.RealSense.SDK.exe"on github

MartyG-RealSense, 2019-03-06:
There are two RSSDK paths installed.  The correct one to find the props is the one I quoted.  The one you linked to stores data related to the SDK, I believe.

The props files have no relation to the current 'RealSense SDK 2.0' on the GitHub.  They are meant for use with the older '2016 R2' or '2016 R3' SDKs.

I have attached to this message a zip of the props folder so you can get the props files.

[props.zip](https://github.com/IntelRealSense/librealsense/files/2935166/props.zip)

RealSense SDK 2.0 uses a different method for setting up a development environment, based around a program called CMake.

bingws, 2019-03-06:
@MartyG-RealSense I want to realize the function of face recognition. The old version of SDK document has related introduction, but the new version of SDK document does not. It's worrying, so try the old version of SDK.

MartyG-RealSense, 2019-03-06:
3DiVi (makers of the popular Nuitrack skeletal tracking software) have a face recognition SDK for $19.99 a month.   It has a free trial version so you can test whether it will work with your SR300.

https://facesdk.3divi.com/

If you would prefer to develop your own face recognition solution then the older '2016 R2' SDK will likely be the best way (face recognition was less stable on the 2016 R3 SDK).

Assuming that you have the SR300's drivers installed and working correctly, the next step will be to download and install the 2016 R2 SDK using the link below, which launches a 1.8 gb download of the installer file in your browser.

http://registrationcenter-download.intel.com/akdlm/irc_nas/vcp/9078/intel_rs_sdk_offline_package_10.0.26.0396.exe

bingws, 2019-03-06:
@MartyG-RealSense I'm not very good at development, so I tried the older SDK first.Thank you.

MartyG-RealSense, 2019-03-06:
I posted a link above for the 2016 R2 SDK if you do not have it already.

bingws, 2019-03-06:
@MartyG-RealSense  I just downloaded it and am installing it.Thank you.

closed 2019-03-06 by bingws