realsenseai/librealsense #3400: The default frame rate of tracking module in realsense-viewer can not be changed

chunyang-zhang, 2019-03-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Realsense T265                     | 
| Firmware Version                   | Unknown version                   | 
| Operating System & Version  |   Ubuntu 16.04                       | 
| Kernel Version (Linux Only)   |  4.4.0-142                              | 
| Platform                                 | x64                                         |
| SDK Version                          |  librealsense 2.18.1                | 
| Language                              |  C++                                       | 
| Segment			        |  Robot/others                         | 

### Issue Description

@dorodnic, hello, I will apply the camera on our moving robot, but the robot moving speed is so fast, so I want to change the frame rate of  tracking module to improve the performance of Visual Odometry , does it work?  How  to change the default frame rate of the tracking module ? Thanks a lot!

- 

- 

-

dorodnic, 2019-03-06:
The default rate is ~260Hz, do you need to increase or decrease it?

chunyang-zhang, 2019-03-06:
@dorodnic , thank you,  I want  to increase frame rate, what should I do?

dorodnic, 2019-03-06:
Hi @chunyang-zhang 
260 Hz is as fast the device can go right now.

chunyang-zhang, 2019-03-06:
@dorodnic  OK, I got it! thank you very much!

dorodnic, 2019-03-06:
Hope it will be sufficient for your use case. Perhaps some kind of prediction algorithm can be added on top? This is what we did for seamless VR HMD tracking.

chunyang-zhang, 2019-03-06:
OK , you did a great job! By the way,  is the source code of prediction algorithm available now?

dorodnic, 2019-03-06:
I think for prediction the team was using one of the existing VR SDKs (either from Vive or Oculus, not sure)
We are planning to release OpenVR wrapper for the T265 in one of the future releases

chunyang-zhang, 2019-03-08:
OK, many thanks, I'm looking forward to your OpenVR wrapper

radfordi, 2019-03-08:
@chunyang-zhang, see #3431 for example prediction code.

closed 2019-04-24 by RealSenseCustomerSupport