realsenseai/librealsense #3401: Point cloud size is not same when captured the same object at different instance

Pratiksha2692, 2019-03-06:
Hi,

Required Info |  
-- | --
Camera Model | {  D435 }
Firmware Version | (05.10.06.00)
Operating System & Version | Linux (Ubuntu 16.04)
Platform | PC
SDK Version | { legacy / 2.. }
Language | {python }
Segment | {3d reconstruction}

### Issue Description
I am trying to capture the .ply file from the export_ply_example.ply. I have added some advance setting as below.
`self.sensors = self.device.query_sensors()
        for sensor in self.sensors:
            if sensor.is_depth_sensor():
                # print(sensor.get_option(rs.option.exposure))
                sensor.set_option(rs.option.enable_auto_exposure, 1)
        advnc_mode = rs.rs400_advanced_mode(self.device)
        depth_table_control_group = advnc_mode.get_depth_table()
        depth_table_control_group.depthClampMin = 0
        depth_table_control_group.depthClampMax = 800
        advnc_mode.set_depth_table(depth_table_control_group)`

I am getting the different point cloud size at different instance when same object is captured by same camera. Why the data points are not same?

Kindly advice.

Thanks in the advance.

RealSenseCustomerSupport, 2019-03-12:
----------------------------------------------

Hi Pratiksha2692,

Can I get this question clarified?
Are you saying that PLY files have different size if you output multiple PLY files while keeping everything (targets and settings) unchanged?

Thanks!

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi Pratiksha2692,

Wonder if you get any udpate? Or do you still need help?

Thanks!

closed 2019-04-16 by RealSenseCustomerSupport