realsenseai/librealsense #3405: Depth Map shows tilted floor

LucidityDesign, 2019-03-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |  11.0.0 | 
| Operating System & Version |   MacOS  | 
| Kernel Version (Linux Only)    |  ---                                         | 
| Platform                                 | Mac |
| SDK Version                            |  2.13.0                          | 
| Language                            |  nodejs                          | 
| Segment			|  others                          | 

### Issue Description
I have a camera mounted parallel to the floor. If I use `nodejs-save-to-disk.js` example, the depth map show a tilt from left to right. The white box is 5cm high, the camera is 35cm above the ground:

[image]

If I use the depth information to draw a depth map. The tilt is much smaller but still noticeable:

[image]

Why can I see this heavy gradient on the floor, even all pixels on the floor have a distance of 34-35cm?

dorodnic, 2019-03-06:
When converting depth to RGB / gray-scale we usually apply histogram equalization. Practically it means that the algorithm **has** to use every color in the color map. When looking at a flat surface it will pick up on the gradient caused by the angle between the plane and the camera and visually emphasize it.

LucidityDesign, 2019-03-06:
But the right side of the image (34cm) has almost the same color as the right side of the box (29cm)

dorodnic, 2019-03-06:
Well, it is not exactly the same color. Histogram equalization will stretch the range of allowed colors over the depth values that are in the scene. So in this case it can be:
[35.1-35.08] black to gray
[35.08-29.1] still gray, since there are no actual pixels in the scene in that range
[29.1-29.02] gray to white
You can control the behavior of the colorizer class from code to disable this effect.

LucidityDesign, 2019-03-06:
How can I config the colorizer with node.js? Unfortunately the node docs are a little limited

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

Hi

Sorry for taking a while to come over to this one.

You may go back to C++ code and play with enabling/disabling histogram equalization and see how it helps:
https://github.com/IntelRealSense/librealsense/blob/master/src/proc/colorizer.cpp#L133

Hope this can be helpful.

Thanks!

RealSenseCustomerSupport, 2019-05-23:
----------------------------------------------

Hi LucidityDesign,

Wonder if there is any update on this one? Do you still need help?

Thanks!

LucidityDesign, 2019-05-24:
Thanks for coming back. I haven't tried to modify the colorizer but ended up drawing my maps via canvas. The results were sufficient for us.

closed 2019-05-24 by LucidityDesign