realsenseai/librealsense #3407: D415 Emitter not turning off and on

Nikhil-Shenoy, 2019-03-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415| 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Windows 7| 
| Platform                                 | PC  |
| SDK Version                            |  2.0 Windows 7                          | 
| Language                            |  C++                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am trying to enable and disable the emitter on my D415 camera, but it is not working. I followed the example from [Controlling the Laser](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser) and other tickets in this repository, but none of them have worked. This is the specific code that I am using to disable the emitter:

`       // Declare RealSense pipeline, encapsulating the actual device and sensors
	rs2::pipeline pipe;

	//Create a configuration for configuring the pipeline with a non default profile
	rs2::config cfg;

	//Add desired streams to configuration
	cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
	cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
	cfg.enable_stream(RS2_STREAM_INFRARED, 1, 640, 480, RS2_FORMAT_Y8, 30);
	cfg.enable_stream(RS2_STREAM_INFRARED, 2, 640, 480, RS2_FORMAT_Y8, 30);

	//Instruct pipeline to start streaming with the requested configuration
	rs2::pipeline_profile prof = pipe.start(cfg);
	rs2::device dev = prof.get_device();
	
	auto depth_sensor = dev.first<rs2::depth_sensor>();
	if (depth_sensor.supports(RS2_OPTION_EMITTER_ENABLED)) {
		depth_sensor.set_option(RS2_OPTION_EMITTER_ENABLED, 0.f);
	}`

With this, I still get the IR pattern visible in my images:

[image]
[image]

What is the correct/best way to disable the emitter temporarily, capture some frames, and then turn the emitter on again?

dorodnic, 2019-03-06:
Hi @Nikhil-Shenoy 
This is the correct way as far as I know. 
Does the stream remain this way, or is this just the next frame after setting the option?
Are you able to control the laser via the RealSense Viewer app?

Nikhil-Shenoy, 2019-03-06:
@dorodnic I reinstalled the SDK because the Realsense Viewer wasn't working. I now can control the laser power and the emitter in the Realsense Viewer, but now no pattern shows up in the captured image. My goal is to capture depth images with the laser on, but IR images with the laser off.

Nikhil-Shenoy, 2019-03-07:
Now I'm not able to turn on the emitter from the Realsense Viewer either

Nikhil-Shenoy, 2019-03-07:
I think I got it now. I read some of the other issues in this repo, and [this one](https://github.com/IntelRealSense/librealsense/issues/1299) mentioned waiting a few frames for the laser to actually turn off. After doing that, I got it to work. Closing.

closed 2019-03-07 by Nikhil-Shenoy