realsenseai/librealsense #3408: Deativate camera laser SR300

AZfrima, 2019-03-06:
| Camera Model                       |  SR300
| Language                               | Python

### Issue Description
I am using 3 cameras to take depth frames. Is there a way to deactivate the laser on each unit so the cameras (without turning the camera off) doesn't interfere with each other ?

MartyG-RealSense, 2019-03-06:
The link below has scripting for controlling the laser using Librealsense1 (Legacy Librealsense) or Librealsense2 (the current RealSense SDK 2.0).

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser

In Librealsense1, the SR300 can only have its Laser Power value altered, as the emitter toggle on-off is for the R200 camera model.  I do not know if Librealsense2 supports laser on-off toggle or just Laser Power like with Librealsense1 .... one of the Intel guys on this forum can clarify that question for you.

Regarding methods of dealing with interference:

Dealing with interference with the SR300 is tricky because unlike the ZR300 and 400 Series models, the 'structured light' projections of the SR300 are susceptible to interference of patterns when they intersect.

Ideally, I would recommend using a process called **enumeration** to only have one camera at a time active - take a snapshot, turn the camera off and activate the next camera in the sequence, repeating the process til you get back to the first camera and the cycle begins again (1, 2, 3, 1).  You have said though that you do not want to turn the cameras off.

Turning the projector off can also have consequences for timing, as there can be a delay of a couple of seconds between turn-off and completion of re-activation.  Constantly turning the projector on and off could also have consequences for the lifespan of the component.  Someone who tried this method also reported that they experienced skipped frames.

Another approach I have considered in the past is a 'lighthouse' one where each SR300 camera is rotated (perhaps with a motor or a turntable) so that although they are all active, only one camera at a time is facing towards the target whilst the projections of the other cameras are facing elsewhere due to their different positions on the rotation cycle.

A problem with turntables would be keeping the cable from tangling up, though you can purchase 'slip rings' for USB cables to help avoid this.

https://www.adafruit.com/product/736?gclid=CjwKCAjww9_MBRAWEiwAlaMJZsqOObasCBP7DVvSy2BEY92ARoHhhDcpS66yZGUivFpPMOsGxodmERoC8_kQAvD_BwE

Yet another approach that Microsoft Research explored with Kinect 1 (which used the same structured light projection as the SR300) was to use motors to vibrate the cameras and break up the interference.

https://www.i-programmer.info/news/194-kinect/3869-shake-n-sense-makes-kinects-work-together.html

A further alternative is to use an external shutter, though these are - according to one user - "ugly and expensive".

RealSenseCustomerSupport, 2019-03-10:
----------------------------------------------

@AZfrima
SR300 is the coded light depth camera. It relies on the projected pattern to calculate the depth. The projector cannot be turned off. You either use one of the 3 SR300 at a time. Or if you have to use 3 cameras for your application, D400 is a better option.

dorodnic, 2019-03-11:
It can be turned off by setting LASER_POWER option to 0, but it is a slow process (you'd need to drop several frames to make sure it took effect)

closed 2019-04-09 by RealSenseCustomerSupport