realsenseai/librealsense #3410: code crash in depth.get_distance()

TouchDeeper, 2019-03-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   |05.11.01.00| 
| Operating System & Version |    Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0                                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.8.1                         | 
| Language                            |  C++                         | 
| Segment			|  Robot                        | 

### Issue Description
Hi,
I wrote a simple program to get distance from the depth frame based on the example of rs-measure, but the code crash when it run to `auto udist = depth.get_distance(u, v);`, and there is no useful message output. I don't know how to fix it. Can you give me some advice. Thanks a lot!
```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2015-2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <librealsense2/rsutil.h>
#include "example.hpp"          // Include short list of convenience functions for rendering
// This example will require several standard data-structures and algorithms:
#define _USE_MATH_DEFINES
#include <math.h>
#include <queue>
#include <algorithm>            // std::min, std::max
#include <thread>
#include <atomic>
#include <opencv2/opencv.hpp>   // Include OpenCV API
#include <mutex>
//PCL
#include <pcl/common/common_headers.h>
#include <pcl/visualization/pcl_visualizer.h>
#include <pcl/io/pcd_io.h>
#include <pcl/filters/passthrough.h>
typedef pcl::PointXYZRGB RGB_Cloud;
typedef pcl::PointCloud<RGB_Cloud> point_cloud;
typedef point_cloud::Ptr cloud_pointer;
typedef point_cloud::Ptr prevCloud;
using pixel = std::pair<int, int>;
//Global variable
int i = 1; // Index for incremental file name
int width = 1280;
int height = 720;
using namespace std;

std::shared_ptr<pcl::visualization::PCLVisualizer> createRGBVisualizer(pcl::PointCloud<pcl::PointXYZRGB>::ConstPtr cloud);
cloud_pointer PCL_Conversion_rect(const rs2::depth_frame& frame, const rs2::video_frame &color, const vector<cv::Point_<u_int32_t>> &vertex,const rs2_intrinsics intr_);

int main(int argc, char * argv[]) try
{
    rs2::align align_to(RS2_STREAM_COLOR);
    // Decimation filter reduces the amount of data (while preserving best samples)
    rs2::decimation_filter dec;
    // If the demo is too slow, make sure you run in Release (-DCMAKE_BUILD_TYPE=Release)
    // but you can also increase the following parameter to decimate depth more (reducing quality)
    dec.set_option(RS2_OPTION_FILTER_MAGNITUDE, 2);
    // Define transformations from and to Disparity domain
    rs2::disparity_transform depth2disparity;
    rs2::disparity_transform disparity2depth(false);
    // Define spatial filter (edge-preserving)
    rs2::spatial_filter spat;
    // Enable hole-filling
    // Hole filling is an agressive heuristic and it gets the depth wrong many times
    spat.set_option(RS2_OPTION_HOLES_FILL, 5); // 5 = fill all the zero pixels
    // Define temporal filter
    rs2::temporal_filter temp;

    // After initial post-processing, frames will flow into this queue:
    rs2::frame_queue postprocessed_frames;

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;

    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH,width,height,RS2_FORMAT_Z16,30);
    // For the color stream, set format to RGBA
    // To allow blending of the color frame on top of the depth frame
    cfg.enable_stream(RS2_STREAM_COLOR, width,height,RS2_FORMAT_BGR8,30);
    // Start streaming with default recommended configuration
    auto profile = pipe.start(cfg);
    auto stream = profile.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();
    auto intr =stream.get_intrinsics();

    cv::Point_<u_int32_t> left_up(540,240);
    cv::Point_<u_int32_t> right_up(740,240);
    cv::Point_<u_int32_t> right_down(740,420);
    cv::Point_<u_int32_t> left_down(540,420);

    // Alive boolean will signal the worker threads to finish-up
    std::atomic_bool alive{ true };
    // Video-processing thread will fetch frames from the camera,
    // apply post-processing and send the result to the main thread for rendering
    // It recieves synchronized (but not spatially aligned) pairs
    // and outputs synchronized and aligned pairs
    std::thread video_processing_thread([&]() {
        // In order to generate new composite frames, we have to wrap the processing
        // code in a lambda
        rs2::processing_block frame_processor(
                [&](rs2::frameset data, // Input frameset (from the pipeline)
                    rs2::frame_source& source) // Frame pool that can allocate new frames
                {
                    // First make the frames spatially aligned
                    data = data.apply_filter(align_to);

                    // Decimation will reduce the resultion of the depth image,
                    // closing small holes and speeding-up the algorithm
                    data = data.apply_filter(dec);

                    // To make sure far-away objects are filtered proportionally
                    // we try to switch to disparity domain
                    data = data.apply_filter(depth2disparity);

                    // Apply spatial filtering
                    data = data.apply_filter(spat);

                    // Apply temporal filtering
                    data = data.apply_filter(temp);

                    // If we are in disparity domain, switch back to depth
                    data = data.apply_filter(disparity2depth);

                    // Send the composite frame for rendering
                    source.frame_ready(data);
                });
        // Indicate that we want the results of frame_processor
        // to be pushed into postprocessed_frames queue
        frame_processor >> postprocessed_frames;

        while (alive)
        {
            // Fetch frames from the pipeline and send them for processing
            rs2::frameset fs;
            if (pipe.poll_for_frames(&fs)) frame_processor.invoke(fs);
        }
    });

//main thread
    while (alive) {
        static rs2::frameset current_frameset;
        postprocessed_frames.poll_for_frame(&current_frameset);
        if(current_frameset)
        {
            auto depth = current_frameset.get_depth_frame();
            auto color = current_frameset.get_color_frame();
            int cloud_width  = right_up.x - left_up.x;
            int cloud_height = left_down.y - left_up.y;
            for (int i = 0; i < cloud_height; i++)
            {
                for (int j = 0; j < cloud_width; ++j) {
                    int index_pcl = i * cloud_width +j;
                    int v = left_up.y + i  ;
                    int u = left_up.x + j;
                    cout<<"u="<<u<<",v="<<v<<endl;
                    auto udist = depth.get_distance(u, v);
                    cout<<"u="<<u<<",v="<<v<<endl;
    }
    alive = false;
    video_processing_thread.join();
    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception & e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
```

ev-mp, 2019-03-07:
Imho, I would check along the following lines:
>cv::Point_<u_int32_t> left_up(540,240);
    cv::Point_<u_int32_t> right_up(740,240);
    cv::Point_<u_int32_t> right_down(740,420);
    cv::Point_<u_int32_t> left_down(540,420);
...
data = data.apply_filter(dec)
...
 int cloud_width  = right_up.x - left_up.x;
 int cloud_height = left_down.y - left_up.y;

Note that the ROI is statically defined presuming that the depth frame resolution is at least 740 pixels wide.  
But then the [decimation filter](https://github.com/IntelRealSense/librealsense/blob/development/doc/post-processing-filters.md) is applied and the resulted image is downsized.   
Failing to  adjust the `left_up`, `right_up` and the other boundaries to the actual image size of the post-processed frame will lead to buffer overrun. The reasonable place to do so would probably be after
> auto depth = current_frameset.get_depth_frame();

TouchDeeper, 2019-03-08:
Hello,@ec-mp
Thank you for your key advice. It is indeed this problem. I just scale down the coordinate value of vertex points, the crash is gone.

closed 2019-03-11 by TouchDeeper