realsenseai/librealsense #3411: pattern noise on IR image

tonyzzzzz, 2019-03-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.10.6 | 
| Operating System & Version |   {Win10  | 
| SDK Version                           | 2}                          | 
| Language                            |  C++                          | 
| Segment			|  machine vision| 

i saw the same pattern noise on my D415, which I just received today. It looks like the noise is caused by the embedded IR LED. i.e., if I cover the internal LED and use an external IR to illuminate the object, the patter noise went away. Can anyone help on this? btw - I did upgraded my D415 to the latest firmware.

D415 out of the box
[image]

Internal IR LED are covered, using ext. IR to illuminate
[image]

btw - I saw there is a similar issue, which has been closed 1 month back..

dorodnic, 2019-03-07:
Hi @tonyzzzzz 
This noise is part of stereoscopic system design. 
You can work without it (by disabling emitter using librealsense APIs) but this will reduce depth quality in some scenarios. 
[Passive / Active Stereo](https://github.com/IntelRealSense/librealsense/blob/master/doc/depth-from-stereo.md#passive-vs-active-stereo)

agrunnet, 2019-03-07:
One other thing to consider is to use the “remove IR” setting when the D415 shows left RGB (not Y8). This will try to reduce the visibility of the IR pattern in the RGB stream, but it won’t remove it completely and it does affect the color balance. Please see the description the the white paper called “Best known methods ...”
Note that this is Not turning off the IR projector. It simply makes It less visible on the RGB stream. Also note that this is left LEFT camera and not the external RGB camera.

tonyzzzzz, 2019-03-08:
@dorodnic @agrunnet thanks. closing issue now.

closed 2019-03-10 by dorodnic