realsenseai/librealsense #3413: Image of depth sensor inwindow inside a Windows Form

jinxzzz, 2019-03-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C# }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am developing an app for my D435 and I am doing it in Windows Form,

How can I show the image of an infrared sensor or depth sensor in a box window inside a Windows form?

Now, my code can show a frame of image in a picturebox, but I want to show depth image and color image in real time.How can I modify the code to show real time image?

Here is my code:
 Pipeline pipeline;//新建管道
                pipeline = new Pipeline();
                Colorizer colorizer;//新建着色器
                colorizer = new Colorizer();

                //开始传送左右成像仪
                var cfg = new Config();
                cfg.EnableStream(Intel.RealSense.Stream.Depth, 640, 480);// Format.Z16, 30
                cfg.EnableStream(Intel.RealSense.Stream.Color, 640, 480, Format.Rgb8);//640, 360, Format.Rgb8, 30);
                var pp = pipeline.Start(cfg);
       

                using (var frames = pipeline.WaitForFrames())
                {

                    var colorFrame = frames.ColorFrame.DisposeWith(frames);//VideoFrame
                    var depthFrame = frames.DepthFrame.DisposeWith(frames);//DepthFrame
             
                    var colorized_depth = colorizer.Process(depthFrame).DisposeWith(frames);

                    var DepthImg = new Bitmap(640, 480, 1920, PixelFormat.Format24bppRgb, colorized_depth.Data);//新建深度图像的内容
                    var ColorImg = new Bitmap(460, 300, colorFrame.Stride, PixelFormat.Format24bppRgb, colorFrame.Data);//新建rgb图像的内容
                    pictureBox1.Image = ColorImg;
                    pictureBox2.Image = DepthImg;

                }

RealSenseCustomerSupport, 2019-03-11:
----------------------------------------------

Hi @jinxzzz,

You can refer to below example to have video streams.
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp/cs-tutorial-2-capture

jinxzzz, 2019-03-11:
> Hi @jinxzzz,
> 
> You can refer to below example to have video streams.
> https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp/cs-tutorial-2-capture

Thanks for your response.
I wrote the winform program by referring to the source code you said.But the source code is written in WPF.WPF is still different from winform in many ways.The winform program I wrote according to this source code does not show depth images. Currently, my program can only display one frame of images, but I do not know how to display real-time depth images.Could you help me check my program and provide me with a small demo to show depth images in picturebox controls?Thank you very much!

snmathur4, 2019-03-15:
Hi, I am also looking for a similar solution? Did you have any luck with this? thanks

RealSenseCustomerSupport, 2019-03-19:
----------------------------------------------

Hi @jinxzzz, @snmathur4,

Try to create a thread to have the depth and color images rendering/refresh in picturebox. A sample code like below.

        void RunInThread()
        {
            while (fThreadAlive)
            {

                    using (var frames = pipe.WaitForFrames())
                    using (var depth = frames.First(x => x.Profile.Stream == Stream.Depth) as DepthFrame)
                    using (var color = frames.ColorFrame)
                    {
                        Console.WriteLine("The camera is pointing at an object " +
                            depth.GetDistance(depth.Width / 2, depth.Height / 2) + " meters away\t");
                        var colorizedDepth = colorizer.Process(depth).DisposeWith(frames);
                        DepthImg = new Bitmap(depth.Width, depth.Height, depth.Stride, PixelFormat.Format24bppRgb, colorizedDepth.Data);
                        ColorImg = new Bitmap(color.Width, color.Height, color.Stride, PixelFormat.Format24bppRgb, color.Data);
                        pictureBox1.Image = DepthImg;
                        pictureBox2.Image = ColorImg;
                    }
            } // end..while
        }

jinxzzz, 2019-03-20:
> 嗨@jinxzzz，@ snmathur4，
> 
> 尝试创建一个线程，使图片框中的深度和彩色图像呈现/刷新。示例代码如下所示。
> 
> ```
>     void RunInThread()
>     {
>         while (fThreadAlive)
>         {
> 
>                 using (var frames = pipe.WaitForFrames())
>                 using (var depth = frames.First(x => x.Profile.Stream == Stream.Depth) as DepthFrame)
>                 using (var color = frames.ColorFrame)
>                 {
>                     Console.WriteLine("The camera is pointing at an object " +
>                         depth.GetDistance(depth.Width / 2, depth.Height / 2) + " meters away\t");
>                     var colorizedDepth = colorizer.Process(depth).DisposeWith(frames);
>                     DepthImg = new Bitmap(depth.Width, depth.Height, depth.Stride, PixelFormat.Format24bppRgb, colorizedDepth.Data);
>                     ColorImg = new Bitmap(color.Width, color.Height, color.Stride, PixelFormat.Format24bppRgb, color.Data);
>                     pictureBox1.Image = DepthImg;
>                     pictureBox2.Image = ColorImg;
>                 }
>         } // end..while
>     }
> ```

Thank you for your answer. It helps me a lot. In addtion , if I want to display a depth image without colorizing, how do I change the code?

freemanlo, 2019-03-20:
A depth image is 16bit depth data. "colorizer" is used to transferred from 16bit to RGB value for better display.

closed 2019-05-13 by RealSenseCustomerSupport