realsenseai/librealsense #3414: Does pyrealsense2 support jetson TX2?

15151874182, 2019-03-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435} | 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    | 4.4.38                                        | 
| Platform                                 |NVIDIA Jetson TX2 |
| SDK Version                            |  2.09 }                          | 
| Language                            |  python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<hello everyone
i have written a python script to get frame and depth data through 'pyrealsense2' on my PC,however, i cant not use 'pip' to install on jetson TX2 directly because of ARM architecture of TX2. Also,there is no source code of 'pyrealsense2' shown on 'pypi' that i cant compile it from source on TX2. So any solutions?>
many thx!

MartyG-RealSense, 2019-03-07:
Advice about using Jetson TX2 with Pyrealsense2 is given on the link below.

https://github.com/IntelRealSense/librealsense/issues/2586

15151874182, 2019-03-08:
> Advice about using Jetson TX2 with Pyrealsense2 is given on the link below.
> 
> #2586

@MartyG-RealSense ,hi, i tried this solution and it works for me, thx!

closed 2019-03-08 by 15151874182